Ultimaker/Cura #15900: Fusion 360 STL file Crashes

lshinampas, 2023-06-24:
### Cura Version

5.4.0-beta 1

### Operating System

Windows 10

### Printer

Anycubic Mega S

### Name abnormal settings

No, only this file crashes.

### Describe model location

Yes

### Describe your model

Fushion 360 file

### Add your .zip here ⬇️

[AI3MEGASPRO_Chicken_Bowl_Feeder_main.zip](https://github.com/Ultimaker/Cura/files/11857488/AI3MEGASPRO_Chicken_Bowl_Feeder_main.zip)

MariMakes, 2023-07-17:
Hey @lshinampas,

Welcome to the UltiMaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

Thanks for the Project File 👍
I took a look at it in Cura 5.4 and saw that I was able to slice your project succesfully without support enabled, because it didn't need support and with the default profile settings. I hope that at least get's you unstuck. 
[image]

I'll have one of the develops take a look when they start resolving slicing crashes. 
Thanks for your report 👍

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic