Ultimaker/Cura #15902: 0.3 layer will not print at 0.3 layer height, it does with layer height 0.299

Linkenelis, 2023-06-24:
### Application Version

5.3.0

### Operating System

Windows 11

### Printer

modifie saphire plus

### Reproduction steps

1. Created a 25 x 25 slab in Fusion 360 and made it 0.3 mm thick (original was more complicated, but I reduced it to this test)
2. Tried to print with "Initial Layer Height" => 0.3 mm
3. Will not print and no show in preview
4. Reduced  "Initial Layer Height" => 0.299 mm
5. Then it prints and shows in Preview
6. Making it 0.3000001 mm thick in Fusion 360
7. It prints  with "Initial Layer Height" => 0.3 mm
8. Feels like something silly like < then instead of <= ?

### Actual results

Will not print and no show in preview

### Expected results

It prints and shows in Preview with "Initial layer height" being less **or equal** to the actual height

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[Body11 Devils Design PETG lw 0.8mm lh 0.3mm if 100.0% ext1 240.0C bed 85C.zip](https://github.com/Ultimaker/Cura/files/11857947/Body11.Devils.Design.PETG.lw.0.8mm.lh.0.3mm.if.100.0.ext1.240.0C.bed.85C.zip)
[failing Body11 Devils Design PETG lw 0.8mm lh 0.3mm if 100.0% ext1 240.0C bed 85C.zip](https://github.com/Ultimaker/Cura/files/11857950/failing.Body11.Devils.Design.PETG.lw.0.8mm.lh.0.3mm.if.100.0.ext1.240.0C.bed.85C.zip)

GregValiant, 2023-06-25:
Thanks for the report and the files.
You have "Adaptive Layers" enabled.  If you turn off Adaptive Layers (it is a single thickness model) then the problem goes away.  You also have the Print Sequence set to One at a Time.  This is a single model slice so that setting is superfluous.

Even with those settings as you have them, the problem does not seem to occur in 5.3.1.  I'm inclined to close this as the problem seems to have been fixed in a later version.  Let us know if you are OK with that.

Linkenelis, 2023-06-26:
Ok Thanks. Works for me.