Ultimaker/Cura #15903: Unexpected behaviour for adaptive layers

rcpoison, 2023-06-24:
### Application Version

5.3.1

### Operating System

Linux

### Printer

Flsun Q5

### Reproduction steps

First time I've ever seen this, usually works like a charm.
Sliced model with vertical curves. 

### Actual results

Sections without any vertical details do not get sliced at max layer height.

### Expected results

As expected, the layers with more vertical curvature result in smaller layer height.
But below/above the cutout, it doesn't slice at max layer height

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[adaptiveLayers.zip](https://github.com/Ultimaker/Cura/files/11858009/adaptiveLayers.zip)
[image]

GregValiant, 2023-06-25:
Thanks for the report.
I don't understand either.  This slice is with the model buried into the build plate.  It would seem that if a feature was causing Cura to keep the layer height at default that this slice should be at the default.  Instead, it's at Max layer height.
[image]

Someone from the Cura team will take a look.

MariMakes, 2023-07-21:
Hey @rcpoison,

I brought your issue up with the team. 
They mentioned that it might be a compounding rounding issue. 
If you increase Adaptive Topography Size, than you see that it's showing more the behavior that you described. 

[image]

We believe we will not work on this anytime soon and will therefore defer this issue.
Since Cura is open source, you or anyone else is more than welcome to work on this issue and create a pull request yourself.

closed 2023-07-21 by MariMakes