Ultimaker/Cura #15904: Smart Hiding of z seam doesn't work when using sharpest corner.

RedeyesXBL, 2023-06-24:
### Application Version

5.3.1

### Operating System

Windows 11

### Printer

Creality Ender S1

### Reproduction steps

Load .stl file, in this case a cat file on the card that came with the printer. 
Make sure to set the z seam alignment to Sharpest Corner, Seam corner preference to Smart Hiding.
Cura does not do a good job at hiding the z seam in the corners. For comparison I have included a screen shot of the result from Creality Print with the same settings. Their smart hiding is a great deal smarter than Cura's at the moment. It would be great if this is something that could be improved for a future release. 

### Actual results

Cura scatters the z seams across the model, regardless of the settings.

### Expected results

Cura should be able to hide the z seams in the corners similarly to the Creality Print software, picture included for comparison. 

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[image]
[image]
[CE3S1_5-Cat.zip](https://github.com/Ultimaker/Cura/files/11858401/CE3S1_5-Cat.zip)

RedeyesXBL, 2023-06-25:
This is same bug as #15882, z seam hiding works as expected with .2 nozzle size, get progressively worse as the nozzle size increases.

MariMakes, 2023-07-05:
Hey @RedeyesXBL,

Welcome to the UltiMaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

As mentioned in the linked issue, we have a known issue where the Hide seam is actually not hidden and can give random results, for now changing the Seam Corner Preference to None will give better results.

This seems to be a duplicate of https://github.com/Ultimaker/Cura/issues/15290.
We have a ticket on our backlog with the intent to improve this behavior.
I'll be closing this issue as a duplicate but you can follow the progress here: https://github.com/Ultimaker/Cura/issues/15290.

closed 2023-07-05 by MariMakes