Ultimaker/Cura #15905: I am printing 4 pieces of a mask, the first 2 sliced fine, the 3rd will not

valicskitten, 2023-06-25:
### Cura Version

5.3.1

### Operating System

windows 11

### Printer

Ender 3

### Name abnormal settings

no abnormal settings

### Describe model location

did not rotate model

### Describe your model

2 of 3 pieces sliced properly

### Add your .zip here ⬇️

[CE3E3V2_HannyaMask_Prop_FDM_Top_Left.zip](https://github.com/Ultimaker/Cura/files/11858621/CE3E3V2_HannyaMask_Prop_FDM_Top_Left.zip)

GregValiant, 2023-06-25:
Thanks for the report and the file.
I was able to slice the project with 5.3.1 and with 5.4beta so I cannot duplicate the problem.  This is in 5.4b.
[image]

This is with 5.3.1 and I didn't have any trouble.
[image]

If you haven't tried it yet - move the model away from the midpoint of the build plate.  Sometimes rotating the model or making a slight adjustment in the Initial Layer Height can work to fool the bug.

The Cura team will take a look.

valicskitten, 2023-06-25:
Thank you! I did move the model and it sliced

GregValiant, 2023-06-25:
I'll go ahead and close this then.  The Cura team has a lot of examples of this behavior.  I hope you understand.

closed 2023-06-25 by GregValiant