Ultimaker/Cura #15906: Pipe Stem

CopperbeardPipeWorks, 2023-06-25:
### Cura Version

5.3.1

### Operating System

Windows 10 Pro

### Printer

HICTOP i3 Custom

### Name abnormal settings

no

### Describe model location

90* Y, 30*Z, Z +1mm

### Describe your model

Yes using S3D

### Add your .zip here ⬇️

I've used this model extensively in Simplfy3D 4.6 with zero issue. Was hoping to use Cura to replace S3D to get better support settings. Crashes any time I try to slice.

[Pipe Stem v8.zip](https://github.com/Ultimaker/Cura/files/11858639/Pipe.Stem.v8.zip)

GregValiant, 2023-06-25:
Thanks for the report and the file.
There is a bug in 5.3.1 that can keep good models from slicing when the model is located at the midpoint of the build plate.  You can try rotating it or moving it.  Sometimes changing the Initial Layer Height by a little bit will fool the bug.
I didn't have any trouble slicing this in 5.3.1 no matter where it was so I'm unable to duplicate the problem.  This is at Z+1mm.
[image]

This is in 5.4beta.  The supports are more robust.
[image]

CopperbeardPipeWorks, 2023-06-25:
I’ve tried printing that way and in the end it doesn’t proved the strength
I need hence the Y 90degree adjustment.

After playing around some more I found it was the 30 degree Z that kept
causing it to glitch

Thank you for the follow up

GregValiant, 2023-06-25:
I'll leave the Bug label on this and the Cura team will take a look.  I haven't been able to get it to "not slice" in any orientation so I'm stuck.

CopperbeardPipeWorks, 2023-06-25:
I’ll play around some more this week to see if I can get it to work.

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic