Ultimaker/Cura #15907: Brim encountered an exception while using it

ABitGinger, 2023-06-25:
### Application Version

5.3.1

### Operating System

Windows 11

### Printer

Anycubic Kossel Liner Plus

### Reproduction steps

1.Use default settings
2.Enable Brim
3.Slice

### Actual results

Abnormal Brim
[image]

### Expected results

Normal Brim
[image]

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[Tape Cutter.zip](https://github.com/Ultimaker/Cura/files/11859430/Tape.Cutter.zip)

GregValiant, 2023-06-25:
Thanks for the report.
See #15838 for more information.  I've made a pull request #15847 to fix this.

I'll go ahead and mark this as a duplicate and close it as it has been addressed and should be fine in 5.4 stable.

You can make the changes locally in the Kossel Linear definition file and the problem will go away.  The file is located in:
"C:\Program Files\UltiMaker Cura 5.3.1\share\cura\resources\definitions" and is named "anycubic_kossel_linear_plus.def.json"
Delete the "machine_disallowed_areas".

You can unzip this and paste the file into the definitions folder.  I've already made the changes to this one.  You will need to over-write the existing file.
[anycubic_kossel_linear_plus.def.zip](https://github.com/Ultimaker/Cura/files/11859754/anycubic_kossel_linear_plus.def.zip)

closed 2023-06-25 by GregValiant

ABitGinger, 2023-06-25:
> Thanks for the report. See #15838 for more information. I've made a pull request #15847 to fix this.
>
> I'll go ahead and mark this as a duplicate and close it as it has been addressed and should be [...]

Thank you for providing the solution!
I used the replacement file you provided to overwrite the original file, but after I re sliced the file I uploaded earlier, this strange phenomenon occurred again, but the location of the occurrence is slightly different:
[image]
I have no deeper understanding of the relevant knowledge and cannot clarify where the specific problem lies. I hope you can further consider how to solve this problem or help me describe and provide feedback on this abnormal phenomenon more clearly.
Thank you deeply again!

GregValiant, 2023-06-25:
That's my bad.  I was moving too fast and forgot about the second file.
Here is another file.  Your printer inherits this definition (the regular Kossel) and it also has Disallowed Areas but they are smaller.  

Do the same with this file as you did to the last one and over-write the existing file.
[anycubic_kossel.def.zip](https://github.com/Ultimaker/Cura/files/11860238/anycubic_kossel.def.zip)

ABitGinger, 2023-06-25:
> 那是我的错。我移动得太快了，忘记了第二个文件。这是另一个文件。您的打印机继承了此定义（常规 Kossel），并且它也有不允许的区域，但它们更小。
> 
> 对此文件执行与对上一个文件相同的操作，并覆盖现有文件。 [anycubic_kossel.def.zip](https://github.com/Ultimaker/Cura/files/11860238/anycubic_kossel.def.zip)

Thank you for your continuous attention!
I have replaced both files.
However, it still cannot function properly and there is a blank area of this shape on the platform:
[image]
[image]

GregValiant, 2023-06-26:
After removing the Disallowed Areas from those two files, I don't see that anymore when I install the Kossel Linear.
That problem appears to be a definition file problem though rather than a Cura bug.
That pattern you see appears to be from the regular Kossel file which has a smaller build surface and smaller disallowed areas.

Maybe you can get some support from AnyCubic on this.

ABitGinger, 2023-06-26:
Okay, I'll try.Anyway, thank you very much for your kind help！