Ultimaker/Cura #15908: CE3E3V2_02 - Top Back Right

whtwlf, 2023-06-25:
### Cura Version

5.3.1

### Operating System

Windows 10

### Printer

Ender 3 v2

### Name abnormal settings

Tree Supports?

### Describe model location

It will slice when moved away from the center of the build plate.

### Describe your model

I have sliced it successfully at other angles. It is not water tight.

### Add your .zip here ⬇️

[CE3E3V2_02 - Top Back Right.zip](https://github.com/Ultimaker/Cura/files/11859756/CE3E3V2_02.-.Top.Back.Right.zip)

GregValiant, 2023-06-25:
Thanks for the report and the file.
This is your file as received:
[image]

I repaired the model with MS 3D Builder.  This is sliced in 5.4beta
[image]

I'm unable to duplicate the problem.   In the meantime have you tried moving the model on the Cura build plate?  Sometimes rotating the model or changing the Initial Layer Height be just a little can help fool the bug.

I'll mark this as a duplicate and the Cura team will take a look.

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic