Ultimaker/Cura #15909: Support angle not correctly calculated (depends on printer selected)

carribeiro, 2023-06-25:
### Application Version

5.3.1

### Operating System

Windows 10

### Printer

Hellbot Magna 2 500

### Reproduction steps

1. Load model with steep angles for slicing.
2. Set layer height and line width for some reference value - let's say, layer height = 0.15mm, and line width = 1.0mm.
3. Select the ENDER 3 PRO profile, enable supports, and check the max support angle. In my case it's 73° (a pretty big inclination, but it works).
5. Now select the HELLBOT MAGNA 2 500, keep all the parameters, and check the angle.  It's different but it shouldn't.

### Actual results

The images below show what happens. The Max Support Overhang is not correctly calculated for the Hellbot, even if the basic parameters that should govern such calculation are the same.

**STL USED FOR THE TEST**

[image]

**BASIC PARAMETERS**

[image]

**SUPPORT OVERHANG ANGLE**

[image]

**THE SAME STL WITH THE HELLBOT MAGNA SELECTED**

[image]

**BASIC PARAMETERS**

[image]

**SUPPORT OVERHANG ANGLE FOR THE HELLBOT MAGNA**

[image]

### Expected results

The MAX SUPPORT OVERHANG for the Hellbot Magna should be the same that the value for the Ender 3, given the same parameters.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[HM2500_Test Cone.zip](https://github.com/Ultimaker/Cura/files/11860196/HM2500_Test.Cone.zip)

The ZIP file contains the test cone. I just squashed it a little bit to increase the angle.

GregValiant, 2023-06-25:
Thanks for the report and the files.
The difference is that the definition file provided by Hellbot doesn't mention the Support Overhang angle so it just inherits the setting as defined in **fdmprinter.def.json** where the actual "value" is not defined.

"support_angle":
                {
                    "label": "Support Overhang Angle",
                          ......
                    "minimum_value": "0",
                    "maximum_value": "90",
                    "maximum_value_warning": "80",
                    "default_value": 50,

Within "creality_base.def.json" file it's different because the "value" is over-ridden and defined as a trig function.
        "support_angle": { "value": "math.floor(math.degrees(math.atan(line_width/2.0/layer_height)))" }

Whether it is the Hellbot, or the Creality printer, the setting is user configurable.
The printer definition files come from a member of the community, or from the manufacturer.  In this case Hellbot and Creality have decided to go different ways with their definitions.  I'll go ahead and remove the bug label and close this.  I hope you understand.

closed 2023-06-25 by GregValiant

carribeiro, 2023-06-26:
Hi @GregValiant . Thanks for the thorough explanation. I'd like to point out though that even if that's not really a "bug" in the sense of the current Cura codebase, it's still something that deserves action - I just don't know how to do it.

Is there any way to ask a partner to provide a fix for the printer profile?

Is there a set of recommendations for printer profiles, something like "best practices", that third parties should adhere to? Is that published anywhere?