Ultimaker/Cura #15915: Slicing fail when using tree supports

mhhayes, 2023-06-26:
### Cura Version

5.3.1

### Operating System

Windows 10

### Printer

Creality Ender-3 S1 Pro

### Name abnormal settings

Attempting tree supports with different parameters

### Describe model location

This issue is confined entirely to tree support meshing. It meshes fine in any other scenario

### Describe your model

Large hole/bridge in center where a support beam will go. Can't orient model hole/bridge perpendicular to build plate. Model made in Inventor Pro and exported from there with high-res STL

### Add your .zip here ⬇️

Strangely enough the crash (I suspect some kind of failure to converge since I had to kill it) happened when I reset tree supports to their default settings.  Simply rerunning the slicer worked. The part itself is simple and fairly unremarkable.

[CE3S1PRO_Chryse_bike_rack_holder_V1.zip](https://github.com/Ultimaker/Cura/files/11863590/CE3S1PRO_Chryse_bike_rack_holder_V1.zip)

MariMakes, 2023-07-17:
Hey @mhhayes,

Welcome to the UltiMaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

The new Cura 5.4 contains the fix for this for a number of slicing crashes. 
I was able to slice your model successfully in 5.4 Stable.
[image]

Could you test if that's the case for you as well? 
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.4.0

If you still run into a crash, please double-check if you shared the correct file and let us know.
I'll take another look at what is going wrong. 👍

MariMakes, 2023-07-24:
Awesome. That's great to hear. Thanks for the update.
I'll close the issue. 👍 

Happy Printing!

closed 2023-07-24 by MariMakes