Ultimaker/Cura #15916: Bridge Settings not being applied to bridge walls.

swd2000, 2023-06-26:
### Application Version

5.3.1

### Operating System

Linux

### Printer

Ender 3

### Reproduction steps

Loaded a 190mm bridging test.
Enabled Bridge settings, including Bridging Wall Speed and Bridge Fan Speed.
Sliced the model.

### Actual results

The bridge walls aren't in the bridge section of the gcode file and use the normal Wall Speed and Fan speed instead of the Bridge-specific variants.

### Expected results

The bridge walls should be within the bridge section of the gcode and use the bridge-specific wall speed and fan settings.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[BridgeTest.zip](https://github.com/Ultimaker/Cura/files/11865880/BridgeTest.zip)

GregValiant, 2023-06-26:
Thanks for the report and the file.
I sliced your project with 5.3.1 and created a gcode.  This is the first "bridge" I come to.
;TYPE:WALL-OUTER
;BRIDGE
M106 S255
G1 F1200 X27.325 Y111.284 E714.69094
;TYPE:WALL-OUTER
M106 S51
G1 F1500 X26.325 Y111.284 E714.76835
G1 F1875 X25.325 Y111.284 E714.83028
**Here is another one:**
TYPE:WALL-INNER
;BRIDGE
M106 S255
G1 F1200 X27.325 Y111.709 E734.60124
;TYPE:WALL-INNER
M106 S51
G1 F1950 X26.325 Y111.709 E734.7123
This is the start of the first skin of the bridge:
;TYPE:SKIN
;BRIDGE
M106 S255
G1 F1200 X26.871 Y122.5 E772.32009

So what I see in the gcode is that the Walls are in their own bridge section and use "Bridge Fan Speed" of 100% and the Bridge Speed of 20mm/sec.  When the bridge ends, the walls revert to their normal speed and fan settings.  Same with the skin.

Post a gcode file that you feel is in error and I'll look at it.  Right now I don't see a problem with the generated gcode.

swd2000, 2023-07-07:
It's probably my misunderstanding. I expected multiple layers of wall to be affected by bridge settings.  I see that the first wall layer in the bridge does use the settings, but then second and third do not and (in this torture test at least), those always sag around and underneath the first bridge wall before they can adhere and cool.  I've printed that test half a dozen times with a variety of speeds and bridge settings.  It's always the second layer of the wall that fails first.  Eventually the walls above adhere to the (good) skin layers inside, but even with settings that have resulted in pretty decent bridges, I cannot avoid a couple of strands of sagged wall.  

When I get a chance I'll hack some gcode (post-processing script) to change the wall speed and fan for the first 3 wall layers in the bridge. I suspect that it will make all the difference.

[bridge.gcode.gz](https://github.com/Ultimaker/Cura/files/11983611/bridge.gcode.gz)

GregValiant, 2023-07-07:
The first layer of a bridge usually goes down pretty good with not much sagging _SO LONG AS_ the lines are far enough apart.
The flat of a 0.4 nozzle is 1mm diameter.  If the lines are too close together then the current line will soften the previous line (because the nozzle is right there) and can cause it to sag since it isn't under tension anymore.
That same sort of thing happens when the second layer goes down.  You can't get any squish because the first layer is very insubstantial and when the first layer re-heats from the second layer, everything wants to sag.  Instead of being flattened into a .20 high x .40 wide rectangle, the extrusion is a cylinder .32 diameter.  It's going to push down the first bridge layer.

Bridges are OK when everything is a straight line (you can't make an arc shaped bridge) and when there are no holes through the overhang (no "stop and go" in bridging).

I've written a post-processor that can change the temperature for bridges.  It's meant for large bore printers but it's something you can play with to see if it helps.
Unzip and put the .py file into the configuration folder.  On Windows it's _...\cura\5.4\scripts_
[BridgeTemperatureAdjustment.zip](https://github.com/Ultimaker/Cura/files/11985237/BridgeTemperatureAdjustment.zip)

masteraxe2, 2023-07-18:
Ah, I had the same misunderstanding. Skin can have up to 3 layers while the wall can only have 1 layer, I see. The walls are giving me trouble though, oh well.

What is happending though is that this (inner wall I think) is ignoring bridging settings. Thoughts?

[image]

GregValiant, 2023-07-18:
Here the inner walls are using the bridge settings.  The "Extra Skin Wall Count" is 1 and it is that "extra" perimeter extrusion (the wide yellow one) that isn't following the bridge rules.
[image]

With the "Extra Skin Wall Count" at 0 things are different.
[image]

Right now there is no way to change the behavior of the Extra Skin Walls for just the bridge layers.

masteraxe2, 2023-07-18:
Thank you very much, that was it!

GregValiant, 2023-07-18:
Since this is "Expected Behavior" I'm going to remove the bug label and change this to a feature request.

The request would be to include Extra Skin Walls in the Bridge Walls, or alternatively, to ignore them for the bridging layers.
Having that one (or more) extrusion go down at full flow isn't good.

@MariMakes can you change the title of this please?  I'm thinking something like "Use Bridge Settings" for "Extra Skin Walls" when appropriate".
I've changed the label.

swd2000, 2023-07-18:
Awesome.  Thank you for your diligence in hunting this behavior down!  It's very much appreciated, and yes, a feature request label is totally appropriate here.

Again, Thanks!

MariMakes, 2023-07-27:
Hey @swd2000,

I've discussed your issue with the team, and it seems like it's something we want to fix. 
We've added a ticket to the backlog with the intent to improve this. 
For internal reference CURA-10873

Thanks for the report! 👍