Ultimaker/Cura #15917: Slicing failed

T-Swebocki, 2023-06-26:
### Cura Version

5.4.0-beta.1

### Operating System

macOS (Sonoma beta)

### Printer

Ender Max

### Describe your model

Some of the elements of this model were sliced before w/o any troubles.

### Add your .zip here ⬇️

https://we.tl/t-ZFrQRw2Rdt

GregValiant, 2023-06-26:
Thanks for the report and the file.
This is another one I can't duplicate.  The project sliced fine for me in 5.4b.  The Cura team will take a look.

[image]

T-Swebocki, 2023-06-26:
Hi,

I am able to slice the project itself, but I am unable to slice it with your extension, that is Tab+ AntiWarping.

Pozdrawiam / Best Regards / Bien Cordialement
Tomasz Swebocki