Ultimaker/Cura #15920: Slicing fails on current 5.3.1 release, works on previous 4.13.1 release

CtrlAltFu, 2023-06-26:
### Cura Version

5.3.1

### Operating System

Windows 10 22H2

### Printer

Snapmaker 2 A250

### Name abnormal settings

Tested with both default and custom profiles, no change.

### Describe model location

Center build plate, rotating doesn't change outcome

### Describe your model

Modular river segment for table top games, meshtools shows no issues

### Add your .zip here ⬇️

[S2A_Wild River-90D-v1.0.3mf.zip](https://github.com/Ultimaker/Cura/files/11869330/S2A_Wild.River-90D-v1.0.3mf.zip)

GregValiant, 2023-06-27:
Thanks for the report and the file.
This won't slice for me either.  I didn't find any errors in the model.
The Cura team will take a look.

CtrlAltFu, 2023-06-27:
Thanks much.

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic