Ultimaker/Cura #15923: unable to slice project

Russianbreadguy, 2023-06-26:
### Cura Version

5.4.0 beta 1

### Operating System

macOS ventura 13.4.1

### Printer

anycubic kobra

### Name abnormal settings

standard settings

### Describe model location

all placed as per normal

### Describe your model

the models are from gambody and have been through QC 

### Add your .zip here ⬇️

[AK_1038-53-55_turbine_connector_L_FDM (repaired).3mf.zip](https://github.com/Ultimaker/Cura/files/11870908/AK_1038-53-55_turbine_connector_L_FDM.repaired.3mf.zip)

GregValiant, 2023-06-27:
Thanks for the report and the file.
It wouldn't slice form me either.  I moved the models and rotated them and they sliced.
[image]
I'll leave this one open for the Cura team and mark this as a duplicate.

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic

Russianbreadguy, 2024-04-03:
Hi,Will do as soon as possible RegardsCalebSent from my iPad