Ultimaker/Cura #15926: Right extruder is using left extruder nozzle temp (IDEX)

paulmendelson, 2023-06-26:
### Application Version

5.3.1

### Operating System

Windows 10

### Printer

Snapmaker J1

### Reproduction steps

I'm using a Snapmaker J1, IDEX. I have PETG in the left extruder, nozzle temp set to 240 C, but I currently have that extruder is disabled. I have PLA in the right extruder, nozzle temp set to 200 C, and it's enabled. I have my part set to the right extruder (PLA). I slice my part, send it to the printer. It heats the right extruder to 240 C instead of 200 C, and I can't figure out why. So I have to manually set it to 200 before it starts printing. Is this a bug? 

### Actual results

Extruder temp is 240 C instead of 200 C.

### Expected results

The right extruder temp will be at 240, even though it's set to 200. It's probably not possible to reproduce this on a single extruder printer. I think it's getting confused in this one case.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

I'm using a Snapmaker J1, IDEX. I have PETG in the left extruder, nozzle temp set to 240 C, but I currently have that extruder is disabled. I have PLA in the right extruder, nozzle temp set to 200 C, and it's enabled. I have my part set to the right extruder (PLA). I slice my part, send it to the printer. It heats the right extruder to 240 C instead of 200 C, and I can't figure out why. So I have to manually set it to 200 before it starts printing. Is this a bug? 

[120mm Fan Connector.zip](https://github.com/Ultimaker/Cura/files/11873121/120mm.Fan.Connector.zip)

GregValiant, 2023-06-26:
Thanks for the report.
I don't have the definition file for the Snapmaker so I only get the model and not the "important stuff".

Cura uses a numbering system for the extruders.  There might be confusion regarding Right#1 Left#2 or should it be Right#2 Left#1.  It really depends on how your firmware treats the numbers.
Are you sure you have it correct?  A disabled extruder should not make it into the gcode.  If you open the Gcode in a text editor you can find lines like T0 (extruder 1) and/or T1 (extruder 2).
Has it been correct for prints that use both extruders?

paulmendelson, 2023-06-26:
Thanks for the response.  I haven't noticed any problems when using both extruders at the same time, or when using only the left extruder. Seems to be the case when I'm using only the right extruder. I should have included the .gcode file. I will attach. Possibly some confusion how the extruders are numbered, like you mentioned. I have no idea offhand.
No idea if the left extruder is actually making it into the gcode. I just know it's setting the nozzle temp to 240, when only the left extruder is set to 240 (and disabled), and the right is set to 200. Possibly some setting I'm missing somewhere, but I don't see it. For example, in cura, under Print Settings, I'm looking at the second tab (2), for the right extruder. If I search for "temperature", nowhere is 240. There is only 200 (and lower). Glancing at the gcode now, which I'm not used to looking at, I see the following lines, but these are just comments I guess? I'm only seeing T1, not T0 when I search the file, so I think the left extruder didn't make it into the gcode. So then, I'm still not sure where the 240 is coming from.  Hmm..

;Extruder 0 Nozzle Size:0.4
;Extruder 0 Material:PETG (1)
;Extruder 0 Print Temperature:240
;Extruder 0 Retraction Distance:1.0
;Extruder 0 Switch Retraction Distance:1
;Extruder 1 Nozzle Size:0.4
[120mm Fan Connector.zip](https://github.com/Ultimaker/Cura/files/11873853/120mm.Fan.Connector.zip)

;Extruder 1 Material:PLA (1)
;Extruder 1 Print Temperature:200
;Extruder 1 Retraction Distance:1
;Extruder 1 Switch Retraction Distance:1
;Bed Temperature:60
;Extruder(s) Used:1

GregValiant, 2023-06-26:
Can you post the gcode file you have trouble with?

Sometimes, if there is a specific callout for a tool in the StartUp Gcode, it can cause problems like this.  Cura lists the starting tool before the StartUp Gcode is inserted so it's possible that the StartUp Gcode can switch to the wrong tool.  Cura doesn't actually read the StartUp gcode so errors can get through.

paulmendelson, 2023-06-26:
Interesting - that could be what's going on then. I figured it's probably some weird little thing I haven't run into before - or just something silly I'm overlooking. Certainly - here is the gcode. 
[fan_gcode.zip](https://github.com/Ultimaker/Cura/files/11874968/fan_gcode.zip)