Ultimaker/Cura #15932: cura not working

Egh2oboy, 2023-06-27:
### Application Version

5.3.0

### Operating System

windows 11

### Printer

Creality CR-10 Smart Pro

### Reproduction steps

was having issues with printing after I went to 5.3.1. So the forums suggested 5.4.1. Downloaded that. It would stop printing for no reason in the middle of a print. I went back to 5.3.0. Was still having the printing issues. So the forums suggested going back to 5.2 Well Now I cant print anything. Says now "print setup disabled" check G Code. I confirmed that the G code is set to Marlin and my printer specs are correct. Ive tried doing searches but now I am frustrated trying to get the program to work. Now I have no idea which version is stable enough to print with. I tried to reset the settings to default but that didnt help either. I am new to this so im trying to learn. but i have some prints that i need to complete and cant now. 

### Actual results

Now I cant print. before it was any part of the print facing the build plate was coming out very stringy. Now it states print setup disabled.

### Expected results

it should allow me to print and function as it was intended to

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

This will not let me save a file. I tried to export it but wont allow me to. Please help dont know what to do

Egh2oboy, 2023-06-27:
Can someone just walk me through this to get my printer back up and running please

GregValiant, 2023-06-27:
Thanks for the report.  I'll try to help...it isn't easy from far away.

Copy any files you want to save from the SD card to your computer and then format the SD card.  You want to make sure the SD isn't the problem and formatting it can clean up some issues if the SD happens to be the problem.

If you do this it will be a brand new install.  All custom profiles need to be set aside.  You want to see if you can get Cura to create a gcode.  Don't worry about the finer points.

In  "C:\Users\...username.....\AppData\Roaming\cura" there should be a folder "5.3".  It has all your custom settings in it.  Rename it to "5.3old".
Uninstall 5.3.1 and 5.3.0.
Install 5.3.1 and then install your printer in Cura.
Slice something (a calibration cube is good) and try to save the Gcode.

MariMakes, 2023-07-05:
Hey @Egh2oboy,

Sorry to hear you are stuck 😖 
Are you still stuck? Did the tip from GregValiant help?
He is right, it's a bit hard to troubleshoot your issue with the information that we have received. 

To me, it doesn't seem like a bug in Ultimaker Cura, but more a question meant for a 3D printing Discord, not an issue for the Cura issue tracker.  Perhaps this can help https://discord.com/invite/3dprinters

Would you still like to keep this issue open or do you agree that we can close it?

closed 2023-07-20 by github-actions[bot]