Ultimaker/Cura #15935: profile error

rponte-BR, 2023-06-27:
### Application Version

5.4.0

### Operating System

windows 11 pro

### Printer

ender 3 S1 plus

### Reproduction steps

import profile

### Actual results

profiles are not loading even installed. Error seen in generic profile.

### Expected results

see all generic profiles

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

error

zcohenld, 2023-06-28:
By the way, this is the same for the Ender 3 S1 regular printer as well as the Plus.

MariMakes, 2023-07-05:
Hey @zcohenld,

Thanks for your report 👍

Could this be a duplicate of https://github.com/Ultimaker/Cura/issues/15771?
This should be resolved in the 5.4 stable release, you can download the Cura version with the fix here: https://github.com/Ultimaker/Cura/releases/tag/5.4.0

Could you let us know if this is a duplicate?
If that's not it, I'll need a little bit more information from you.
Could you perhaps share some screenshots of what you are seeing?

closed 2023-07-20 by github-actions[bot]