Ultimaker/Cura #15937: 5.4.0 b1 - Tree supports don't support small starting elements

tlhintoq, 2023-06-28:
### Application Version

5.4.0b1

### Operating System

Win10x64pro

### Printer

Any

### Reproduction steps

Using Ultimaker's own tree support model on Thingiverse:
https://www.thingiverse.com/thing:6069926

Can not in any way shape or form get it to support the goatee and tenticles coming off the head.  All start in mid air, and so the rest of the head is just going to fail.

If there is some setting "minimum size to support" or something I can't find it or know the phrase to search.
[image]

1. It would be nice if a posted test part didn't need to be repaired when downloaded.  I think that's counterintuitive for people expecting a working test part.
2. I'm stunned there was no complete Cura test project .3mf with this upload for people to have a known good working starting point.  For any printer, an Ulitimaker, and Ender3  something to open and look at.

### Actual results

Supports should be supporting things.

### Expected results

Supports _not_ supporting 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Solution file is 36mb and limit here is 25, so can't even include a compressed .3mf of your own test project.

GregValiant, 2023-06-28:
Thanks for the report.
A problem is that there are 73 support settings and a lot of them need to be tweaked for this model.  Start out by selecting the "Custom" settings.  To the right of the "Settings Search Box" is a button with three lines on it.  It's is the Setting Visibility Tool.  Set the visibility to "All".

I think the main ones you are looking for are "Minimum Support Interface Area" and "Support Interface Density".  "Z overrides XY" should be checked.  A high "Support Interface Density" (like 60%) may be required.  Those chin whiskers are tough to support.
(I think it would have been better to include a project file with the model so users could see how it was done.  This model is "Advanced Support Design" and not something a newer user would be comfortable with.)

This is in 5.4beta1 and although I didn't spend much time on it the chin is supported.
[image]

tlhintoq, 2023-06-30:
That's for the fast reply and the helpful values.
I already had Z before XY.
Did support interface density between 20 and 100 in 10% jumps.
I had not previously tried dropping the angle 30° but tried after your suggestion: No change.
I commend you getting the whiskers to support but it looks like you have a complete encasement to do it.
Its certainly nothing as friendly as what Ultimaker displays on this test part.
[image]

> This model is "Advanced Support Design" and not something a newer user would be comfortable with.
Agreed.  I would have figured about 5 years with user Cura would have made this less of a nightmare and less of a noobie situation for me.

> (I think it would have been better to include a project file with the model so users could see how it was done.
That's what I'm saying.

ThomasRahm, 2023-07-07:
Can you check the `minimum support area` setting?
This could be expected behavior if it is larger than 0. 
If it is 0 please attach your profile, so that i can try to replicate it.