Ultimaker/Cura #15938: Tree support non-deterministic (5.3.1)

cmerrill, 2023-06-28:
### Application Version

5.3.1

### Operating System

Windows 11

### Printer

Anycubic Kobra

### Reproduction steps

1. Added my model and added support blocker
2. Attempted to slice with tree supports enabled
3. Tree supports did not support a portion of the model
4. Enabled/disabled a random setting (retract at layer cahange) in order to enable slicing, and rotated view
7. sliced
8. This time the tree support supports all needed parts of the model

[image]
[image]

### Actual results

Slicer produces different support with the same model.

### Expected results

Slicer should produce consistent support with the same input model.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

The project I am talking about:
[lock_mount_example.zip](https://github.com/Ultimaker/Cura/files/11889381/lock_mount_example.zip)

Log file:
[cura.log](https://github.com/Ultimaker/Cura/files/11889387/cura.log)

GregValiant, 2023-06-29:
Thanks for the report and the files.
When I sliced the project the supports look as you described.  That large overhang was unsupported.
I changed to your "Suggested ABS" setting profile and it looks like this:
[image]

So this appears to be a settings issue rather than something wrong in Cura.  If you go over the Support settings for that ABS profile and transfer them over to the PETG profile I think you will get the results you want.
I'll leave the bug label on this for a bit and I'll leave it open so you can respond, but this does look like a settings problem.

GregValiant, 2023-07-01:
I'll go ahead and remove the bug label and close this.  If you feel that is premature let me know and I can re-open this.

closed 2023-07-01 by GregValiant

DerGenaue, 2023-07-09:
Should be solved by the new implementation in Cura 5.4 btw