Ultimaker/Cura #15944: Loading of files limited to single file rather than a group

leotilson, 2023-06-28:
### Application Version

5.4.0-beta.1

### Operating System

Windows 10

### Printer

Creality CR-10 Max

### Reproduction steps

Tried to load stl files into Cura

### Actual results

Only able to select one file at a time. The single file selected loads successfully.

### Expected results

Previously, was able to select multiple files at once using a combination of shift/control and mouse click.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

N/A

GregValiant, 2023-06-28:
Thanks for the report.
I had noticed this.  It worked as you described in 5.3.1 but not in the 5.4 beta.  I'll leave the bug label on this but it might be a new "expected behavior".

leotilson, 2023-06-28:
Thank you for the individual response Greg

All of Cura is very much appreciated - a magnificent program.

Regards

Leo

MariMakes, 2023-07-05:
Hey @leotilson,

Welcome to the UltiMaker Cura Github 🚀 
Thanks for your report and the magnificent compliment 😊👍

This seems to be a duplicate of https://github.com/Ultimaker/Cura/issues/15879
We have a ticket on our backlog with the intent to improve this behavior.
I'll be closing this issue as a duplicate but you can follow the progress here: https://github.com/Ultimaker/Cura/issues/15879

closed 2023-07-05 by MariMakes