Ultimaker/Cura #15945: Tree Support and brim overlap

santiagomoneta, 2023-06-28:
### Application Version

5.4.0 beta

### Operating System

mac os

### Printer

ender 3 v2

### Reproduction steps

import viking model (https://www.thingiverse.com/make:1107990)
enable brim
enable tree support
set layer to 0.2mm
slice 

### Actual results

layer 1 have the complete brim
layer 2 start the actual base ... on top of the brim

### Expected results

the brim respect the model on layer 1

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

GregValiant, 2023-06-28:
Thanks for the report.
I get errors when trying to download that file but it looks like the bottom of the model isn't flat.  You can see that on the first layer there are small points of the model that start around the base.  The rest of the base is floating above the build plate and the brim is going around the points.
Try setting the model to -0.1 in the Z and see if it makes a difference.
What I see here is a problem with the model rather than a bug.  A Project File would help.
I'll remove the bug label but leave this open so you can respond.  I hope you understand.
You could also try making the base of the model flat using something like MS 3D Builder or Blender.

GregValiant, 2026-09-10:
I'll close this.

closed 2026-09-10 by GregValiant