Ultimaker/Cura #15946: Infill Wipe Disappears when Infill Density above 70%

adamrumjahn, 2023-06-28:
### Application Version

5.3.1

### Operating System

Windows 10

### Printer

Custom

### Reproduction steps

1. I sliced with Infill Density 70%, Infill Wipe Distance 0.4mm  - Works fine (see wipe at end of infill lines below, note that shell is hidden)
[image]

2. then I sliced with Infill Density 71% (anything above 70% creates this error), Infill Wipe Distance 0.4mm - Wipe Disappears... (notice that wipe has disappeared from infill lines)
[image]

I'm not sure what is causing this, but it would be really helpful to keep the wipe regardless of the infill percentage.

### Actual results

Wipes disappear with any infill density above 70%.

### Expected results

Wipe moves should remain regardless of infill density.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

I've included the project file here.  Just try slicing with 70% infill density and above to reproduce the issue.
[cure531-wipe issue.zip](https://github.com/Ultimaker/Cura/files/11895131/cure531-wipe.issue.zip)

GregValiant, 2023-06-28:
There are so many errors in your project file that it won't slice.  A print temperature of 450° is a tad high.

adamrumjahn, 2023-06-29:
It's a custom printer. Feel free to adjust the temperatures so that it will
slice and then try again. The temperatures don't affect the wipe bug anyhow.

adamrumjahn, 2023-06-29:
In an infill around a circumference, it's selective about where to apply the infill wipe.  See below
[image]
The issue that I'm having with the above scenario is that some areas create gaps between the infill and the inner walls.  The wipe fixes this, but only in some areas, not others.

If I try to use a concentric infill, I would also expect that the wipe would happen into the infill itself, but it doesn't, it just skips the wipe completely.  See below:
[image]
The issue that I'm having with the above is that there is a gap between every consecutive concentric infill start/stop line.