Ultimaker/Cura #15948: Cura 5.4.0-beta.1

Crylia, 2023-06-28:
### Application Version

5.4.0-beta.1

### Operating System

Linux

### Printer

-

### Reproduction steps

Try to start application

### Actual results

Not starting

### Expected results

Starting

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

When trying to launch the appimage I get this error.

PyInstaller/loader/pyimod02_importers.py:352: DeprecationWarning: the imp module is deprecated in favour of importlib and slated for removal in Python 3.12; see the module's documentation for alternative uses

N6REJ, 2023-07-01:
i just get a hard crash.

jellespijker, 2023-07-01:
- Is this with the modern AppImage or legacy?
- Are you using Ubuntu 23.04?
  - What is the outcome of: `uname -a`
- What do the logs say? They can be found at: `~/.local/share/cura/5.4/cura.log`

N6REJ, 2023-07-01:
in my case its windows 10 x64 pro

jellespijker, 2023-07-01:
`linux.AppImage` are for Linux operating systems. Use either the `win64.exe` or `win64.msi` installers.

N6REJ, 2023-07-02:
no, I used the win64.mis and the win64.exe with same results

N6REJ, 2023-07-02:
[image]
I assume its sent numerous crash reports during my attempts

jellespijker, 2023-07-02:
please open up a new issue for this and make sure you add the logs.

jellespijker, 2023-07-02:
@Crylia 

Is this with the modern AppImage or legacy?

Are you using Ubuntu 23.04?

What is the outcome of: uname -a

What do the logs say? They can be found at: ~/.local/share/cura/5.4/cura.log

Crylia, 2023-07-02:
`Pop!_OS 22.04 LTS`
`Linux chunchumaru 6.2.6-76060206-generic #202303130630~1685473338~22.04~995127e SMP PREEMPT_DYNAMIC Tue M x86_64 x86_64 x86_64 GNU/Linux`
There are no logs

jellespijker, 2023-07-02:
@Crylia 

Also not the error output, which can be found at: ~/.local/share/cura/

Crylia, 2023-07-02:
Uh I redownloaded the appimage as I deleted it, and now it works just fine?

jellespijker, 2023-07-02:
good to know will close this issue

closed 2023-07-02 by jellespijker