Ultimaker/Cura #15951: Slicing failed with support blocker

Champ1800, 2023-06-29:
### Cura Version

5.4.0-Beat.1

### Operating System

Windows 11

### Printer

Ender 3 V2 running klipper

### Name abnormal settings

I added support blocker & scaled the x axis to 30mm

### Describe model location

Yes, if I rotate it 90° it will slice ok.

### Describe your model

Yes without any support blocker

### Add your .zip here ⬇️

[Failed slicing.zip](https://github.com/Ultimaker/Cura/files/11904714/Failed.slicing.zip)

I added a narrow support blocker to the top section & it failed to slice.
It does slice it I rotate the model 90, 180 or 270°C.
Also after moving the support blocker around it would then slice in its original position.

MariMakes, 2023-07-05:
Hey @Champ1800,

Thanks for your report 👍

I can reproduce your crash in Cura 5.4. And, like you, it's resolved when I move the model on the buildplate. 
[image]

This is an excellent testcase for testing the Slicing Errors. 
I'll have one of the developers take a look.

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic