Ultimaker/Cura #15956: Sonic Hedgehog Right Hand - Slicing Failure

gugamc, 2023-06-29:
### Cura Version

5.4.0 Beta 1

### Operating System

Windows 11 Pro 22H2

### Printer

Ender S1 Pro

### Name abnormal settings

Nope

### Describe model location

Nope

### Describe your model

Yes with other parts of Sonic, however the file alone does not slice.

### Add your .zip here ⬇️

[Sonic Right Hand.zip](https://github.com/Ultimaker/Cura/files/11911206/Sonic.Right.Hand.zip)

GregValiant, 2023-06-30:
Thanks for the report and the file.
When I import the STL and the support is part of the model it slices fine.  When I split the model using Mesh Tools and try to print the support "As Support" it fails and I can't get it to slice.
There have been other reports of this behavior so I'll mark this as a duplicate.  I'll leave it open as the behavior is a bit different and the Cura team may like to see this example.

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic