Ultimaker/Cura #15958: concentric top and bottom layers cause error

techdeasy, 2023-06-30:
### Cura Version

5.4.0 beta 1

### Operating System

win 11

### Printer

anycubic mega pro

### Name abnormal settings

concentric top and bottom

### Describe model location

centered 

### Describe your model

first time trying to print

### Add your .zip here ⬇️

[AI3MSPRO rad_gewindeeinsatz_fixed Generic TPU 95A lw 0.4mm lh 0.2mm if 30.0% ext1 215.0C bed 75.0C.zip](https://github.com/Ultimaker/Cura/files/11915707/AI3MSPRO.rad_gewindeeinsatz_fixed.Generic.TPU.95A.lw.0.4mm.lh.0.2mm.if.30.0.ext1.215.0C.bed.75.0C.zip) 

printing in pla not tpu, too lazy to change

GregValiant, 2023-07-01:
Thanks for the report and the file.
It wouldn't slice for me until I moved the model away from the midpoint of the Cura build surface.
You can see that the "Slicing Failed" warning of the previous attempt had not dismissed yet.  
[image]

I'll mark this as a duplicate for the Cura team.

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic