Ultimaker/Cura #15959: Slicing part individually succeeds, but when added to print bed, layers are missing.

madhephaestus, 2023-06-30:
### Application Version

5.1.1

### Operating System

Ubuntu 22.04

### Printer

Just a slicer problem

### Reproduction steps

1. Open the part as a single part
2. slice normally
3. open print bed containing part (a single stl composite of STL's)
4. Slice new STL, layers are missing from part

### Actual results

Sections of a part are missing.

### Expected results

the part in the print bed is sliced the same as the part stand alone. 

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

Zip of problematic files: https://github.com/Halloween2020TheChild/GroguMechanicsCad/releases/download/0.4.6/hephaestus-archive-0.4.6.zip

Stand alone file that works is: BaseMountStandard6dofArm1.stl
Print bed dumb-unioned with missing layers: Print-Bed-0.stl

[image]

Asterchades, 2023-07-01:
Missing layers are generally the result of model errors. In this case I suspect it to be the result of cumulative errors - a single faulty model might still be within the realm of Cura's correction facilities, but once you add more to the bed you're now moving beyond what it can accommodate for as the errors interact with each other.

Indeed these models are reporting as faulty per 3D Builder, and if I export a copy of the full bed after repairs it slices correctly. Importing the original into SketchUp doesn't show anything immediately obvious, but on closer inspection there is a _lot_ of geometry which has co-linear edges that aren't actually shared, leaving the models non-manifold and with floating geometry. This isn't helped by there being a lot of excessive geometry, to the point many of those "flat" surfaces aren't actually flat (which is reflected in the slice).

To that end I'd like to ask you to run these models through a correction tool. Unfortunately I'm unaware of any which function under Linux as I don't use the OS, but if you have access to a Windows 10 or 11 machine the free MS 3D Builder tool is really quite exceptional at this.

madhephaestus, 2023-07-01:
is there a way to see the errors in Cura?

madhephaestus, 2023-07-01:
Hmm, can you show me where you calculate the planar Epsilon used to clip the surfaces? It sounds like Cure has the same bug my CAD engine had: https://github.com/miho/JCSG/pull/64/files

I am able to perform CSG operations with these models, and as a result the CAD engine is seeing them as manifold. When you say "colinear edges", by what Epsilon value?

Asterchades, 2023-07-02:
> is there a way to see the errors in Cura?

Not really. Cura will show some errors as polkadot patterns - cyan/magenta or blue/green in the regular Prepare screen, or as red marks in the X-Ray view under Preview. I've never been entirely clear on what each is meant to represent, though I'm pretty sure the cyan/magenta pattern is superfluous internal geometry.

The best I can see with these models is to change to the X-Ray view on the Preview tab. Then if you zoom or move the camera you'll see blink-and-you-miss-it red speckles on some of the models, particularly the flat top surfaces. Understandably something very easy to miss as even I wouldn't have spotted it if I wasn't looking specifically for it. I suspect these are just too detailed and the faulty geometry is being "suppressed" by the surrounding good geometry.

> Hmm, can you show me where you calculate the planar Epsilon used to clip the surfaces? It sounds like Cura has the same bug my CAD engine had: https://github.com/miho/JCSG/pull/64/files
>
> I am able to perform CSG operations with these models, and as a result the CAD engine is seeing them as manifold. When you say "colinear edges", by what Epsilon value?

This sort of thing is unfortunately well and truly beyond my level of expertise. While I'm reasonably well versed in the user-side portions of Cura, the back-end and the math involved may as well be magic. However as I've seen this surface distortion before and I know it's been under investigation I'll give @MariMakes a ping so she can hopefully direct it to one of the developers who can better understand the implications.

To determine the co-linear edges I used SketchUp Make 2017, so I'm relying on its calculations being accurate. It was simply the distinction of heavier lines (edges) against lighter ones (joints), with my being able to remove some heavier lines and only destroy one poly in the process rather than both.

madhephaestus, 2023-07-02:
I can not seem to reproduce the error display in Cura. Can you show me with a screencap? 

I have run these meshes through MeshLab and it has not shown any errors. 

I was reading the original STL file format specification here: http://www.fabbers.com/tech/STL_Format and i do not see any mention of co-liner edges as an error.  The STL's above adhere to the vertex-to-edge rule, so i am having a hard time determining what the error in the mesh is? And if the mesh is not in error, then doesn't the reported behavior represent an actual bug in Cua's behavior? Especially given the model slices well individually, but the slice fails when the part is in a print bed.

madhephaestus, 2023-07-02:
the only error message WRT the slicing:

```
2023-07-02 12:28:36,448 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [111]: [Backend] [WARNING] Mesh has overlapping faces!
2023-07-02 12:28:36,457 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [111]: [Backend] [WARNING] Mesh has disconnected faces!
```

Asterchades, 2023-07-02:
It really is blink-and-you-miss it stuff when viewed in Cura. This is about the most obvious I've seen, and calling it "obvious" is stretching the meaning of the word quite a bit.

[image]

The co-linear edges I'm talking about (and at this point it's safe to assume that's not exactly the right term) are easier to identify in SketchUp. By default (and I'm not even sure if it can be changed) it will use thinner lines to show when two surfaces are joined (even if not co-planar) and thicker ones to show an edge or significant angle. The line pictured here heading towards the upper left is one of the thicker ones:

[image]

Given the obscene level of superfluous geometry in this the possibility exists that it's a double edge, true. So I used the move tool to shift the line and wound up with this:

[image]

The burgundy that's now visible is my back face colour. While the edges in the other directions stayed connected, this one just outright split the model as a result of the geometry on both sides not sharing a common edge. They do have edges in the same place, yes, but they aren't the same edge.

Also, while trying to find an example of the above edges I came across this: straight-up Z fighting of two surfaces facing opposite directions trying to exist in the same space.

[image]

The entire object that went missing on my build plate (rear left - the taller object with the round top) also appears to have been described twice. Individual polygons can be removed to reveal the same surface underneath, though they are all ultimately attached to one another.

As mentioned, running it through 3D Builder results in all of these errors inconsistencies disappearing. The Z-fighting parts are removed entirely along the edge of the recess leaving a clean "round" hole and the geometry is properly "stitched" together. However even it appears to be struggling with some of it as at least two of the hole openings do this (and there's possibly other "damage" done during the repair):

[image]

Unfortunately I cannot find a way to get a list, or even a count of the repairs in 3D Builder.

Again, the "how" and "why" of all the above is beyond my understanding. I can only point out the "what" of it all - though if 3D Builder's repairs are anything to go by, part of it may be that trying to put vertices part-way along an edge results in the surface no longer being fully cohesive.

MariMakes, 2023-07-05:
👋 I've been summoned 😊 

What an amazing analysis @Asterchades. 😍 

To me, it doesn't seem like a bug in Ultimaker Cura, since the mesh is faulty and Cura is not built to alter meshes in this way, that's what CAD software is for.

This is where Cura checks if a mesh is watertight https://github.com/Ultimaker/Cura/blob/1a4a3c27b98294d1cb29c3829041dd06bc7df0d7/plugins/SolidView/

Was that the information you are looking for? 
If that's not it, what's the best way that I can help here?

madhephaestus, 2023-07-05:
Thank you for popping in @MariMakes !

 I was looking for the value used to determine if 2 edges are just close, or manifold. It should be some small number, 10^-6 or something. I want to make sure my triangualtion algorithm is using the same Epslion values for how close is "the same". In floating point land its always not exact, so i was looking to find out precisely how exact your algorithm expects features to be.

madhephaestus, 2023-07-06:
The original issue of slicing failing was the result of a model being included twice, exactly on top of itself. The surfaces were perfectly allinged, so entirely invisible in all Mesh analysis tools.

@Asterchades thank you for your deep dive into the errors, that helped get me on the right track!

@MariMakes I am not closing this right away hoping to get an answer  on the Epsilon value used in cure. It is low priority now that slicing is working after my print bed bug was resolved.

Asterchades, 2023-07-06:
Glad I was able to help out a bit, even if not directly. I'll swap this one over from a bug to a discussion - I have a feeling there could be more to this that people smarter than me can make use of.

GregValiant, 2026-09-10:
I'll go ahead and close this.

closed 2026-09-10 by GregValiant