Ultimaker/Cura #15962: Slicing fails when using custom supports with large models

JarydBaker, 2023-06-30:
### Cura Version

5.4.0-beta.1

### Operating System

Windows 11

### Printer

Ender 3 V2

### Name abnormal settings

Print 1 at a time, gantry height

### Describe model location

No rotation allows it to print

### Describe your model

3 largish parts with custom supports. Without custom supports, the mesh is successful.

### Add your .zip here ⬇️

[CE3E3V2_Jabberwock body_wings.zip](https://github.com/Ultimaker/Cura/files/11918547/CE3E3V2_Jabberwock.body_wings.zip)

The model will mesh without custom supports, but fails when using custom supports.

GregValiant, 2023-07-01:
Thanks for the report and the file.
This won't slice for me either.  If I change the support cylinders to "Print as Normal Model" then it slices.
The Cura team will take a look.

MariMakes, 2023-07-05:
Hey @JarydBaker,

Welcome to the UltiMaker Cura Github 🚀 
Thanks for your report 👍

I recognize this issue. 🕵️‍♀️
This seems to be a duplicate of https://github.com/Ultimaker/Cura/issues/15836, where using custom supports would result in a slicing crash. 

It's resolved in the 5.4 stable release, you can download the Cura version with the fix here: https://github.com/Ultimaker/Cura/releases/tag/5.4.0
I'll close this issue since it's a duplicate and resolved.

closed 2023-07-05 by MariMakes

JarydBaker, 2023-07-05:
Hi @MariMakes, thanks for the update, that is great to hear :).