Ultimaker/Cura #15963: Printers settings not getting saved

rajinder-yadav, 2023-06-30:
### Application Version

5.3.0

### Operating System

openSUSE Linux KDE 

### Printer

Tevo Tornado

### Reproduction steps

1. Salect: Manage printer
2. Click on "Machine Settings"
3. Deleted all start and stop gcode in Printer Tab
4. Close Setting dialog 
5. Exit application and reload

(*) There might be other setting changes not getting saved correctly.

### Actual results

When you check printer settings the gcode is still there.

### Expected results

start / stop gcode should be deleted

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

nothing

GregValiant, 2023-07-01:
Thanks for the report.
That was a known bug in early 5.x versions of Cura.  The Machine Settings dialog in particular had issues.
You can try updating to 5.3.1.  I use Windows so I can't duplicate the problem but I have not heard of this bug in 5.3.1 or 5.4beta.  Let us know if the problem persists with a later version.

rajinder-yadav, 2023-07-04:
The bug seems to have gone away? I was using v5.3.1. I did end up correcting directory/files permission for group. I am going to close it off, as I do see my changes being saved.

closed 2023-07-04 by rajinder-yadav

GregValiant, 2023-07-04:
Thanks for getting back and glad it's fixed.