Ultimaker/Cura #15968: Extruder number mismatched

JLombardia, 2023-07-01:
### Application Version

5.4.o beta 1

### Operating System

Windows 11

### Printer

Sovol SV04

### Reproduction steps

1.- Slice in Dual Mode
2.- Select parts for extruder 1 / left / T0
3.- Define supports with extruder 2 / right / T1
4.- Slide

### Actual results

The gcode swap the extruders and use E2 for parts and E1 for supports.
Comparing gCode from same part and setup in Cura 5.3.1 the command T0 and T1 are swapped, as temperatures and other parameters. 

### Expected results

All the T1 must be T0 and all the T0 must be T1.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

nothing else

GregValiant, 2023-07-01:
Thanks for the report.
I don't have the "SV04 in Dual Mode" in 5.3.1.  It is present in 5.4beta.
How did you install the printer in 5.3.1?

When I save a project file in 5.4b using the SV04 in Dual Mode, and try to open it in 5.3.1 - 5.3.1 crashes.

JLombardia, 2023-07-01:
I downloaded the profiles from the Sovol website. https://drive.google.com/file/d/1f52wbuQpgRlsocqLN5z08jPrf_gwLMDB/view?usp=share_link

To use a profile from 5.4 in 5.3, you must export them and import in 5.3 after install the machine.

Copy the 2 folders inside Cura config folder, open Cura 5.3.1 and you can add the SV04 machines.
And yes, I can select different temperatures and extrusion in each extruder. I use the left with PLA+ 220° or Nylon 250°, and the right for supports with PVA 190°.
The problem with 5.4 beta is that print the parts with the PVA at 220° and print the supports with the PLA at 190°. And it happens because in the Gcode all T0/T1 instructions are swapped.

GregValiant, 2023-07-01:
This was a head scratcher.
The bottom line is that it appears to be a Sovol definition file problem rather than a Cura bug.
Your 5.4b definition uses the extruder files: 
"0": "sovol_sv04_dual_mode_extruder_0"
"1": "sovol_sv04_dual_mode_extruder_1"
but the 5.3.1 definition uses extruders:
"0": "SV04_extruder_0"
"1": "SV04_extruder_1"
there are also extruders:
"SV04_extruder_left.def.json"
"SV04_extruder_right.def.json"

Because the "Copy" and "Mirror" definitions are slightly different than the "Dual" definition which extruder file is loaded appears to make a difference.

So I think you need to go back to Sovol and ask them to clarify why the two definition files use different extruders.  Somehow, someway, the extruder definitions appear to be causing the problem.  It might be as simple as changing the printer definition files so that they each use the same pair of extruders.  You might also have to go over the Mirror and Copy definition files and figure out what is going on with them as well.

I'll remove the bug label from this but I'll leave it open for a while so you can respond.  I hope you understand.  The "T0 should be T1" problem has not come up here before and there are numerous dual extruder machines (including the UltiMakers) that would be affected if this were a Cura bug.

JLombardia, 2023-07-01:
In Cura 5.3 I'm using the downloaded Sovol definitions and work fine.
In Cura 5.4 I'm using the definitions included with Cura. With this the slide have the confusion. Seems the problem is with the definitions that Sovol gave to Ultimaker.

GregValiant, 2023-07-01:
I think it's a definition problem.  I go through a lot of printers here and I've never seen this confusion with the Geeetech's, multi-extruder Enders, or the Ultimaker printers.  I don't remember this being a problem with the Sovol's either until this report.
It might come back to a Cura problem, but I don't really see how.  If the definition files are right then the gcode is right.

There was a line in the movie "Short Circuit" - _"It just runs the software!"_.  Mostly.  But sometimes strange things seem to happen.

GregValiant, 2024-04-05:
This is a definition problem.  I'll go ahead and close this.  Let me know if you feel that is premature.

closed 2024-04-05 by GregValiant