Ultimaker/Cura #15969: Every File Is Failing To Slice

Metroidfreak33, 2023-07-01:
### Cura Version

5.4.0 beta1

### Operating System

Windows 11

### Printer

Sovol SV06 Plus

### Describe model location

Center of plate

### Describe your model

A flat round disk

### Add your .zip here ⬇️

[Base.zip](https://github.com/Ultimaker/Cura/files/11925934/Base.zip)

Everything I attempt to slice results in a slicing error. This is the simplest file I'm trying to print and it still fails. These all slice in 5.3.1 without issues.

Metroidfreak33, 2023-07-01:
Uninstalling all 3rd party extensions did not resolve this.

GregValiant, 2023-07-01:
Thanks for the report.
When I opened your project file the printer would not import.  I had to install an SV06 in order to check.  That should not have to happen.
This might be a problem with the Sovol definition file.  I certainly haven't run into the problem with any of the other printers I have installed.
See if un-installing Cura and re-installing it makes a difference.  There have been a few reports lately of problems with the Sovol definition files and extruder files.

Metroidfreak33, 2023-07-01:
I uninstalled every version of Cura I had since I couldn't figure out how to reinstall the new version without it pulling in the other versions configurations and profiles. Looks like a full wipe and inputting the printers from scratch as Custom worked. Other settings that were broken before are now working as well, so there was definitly something wrong on my end. Thanks for the reccomendation.

closed 2023-07-01 by Metroidfreak33

GregValiant, 2023-07-02:
I'm on a Win10 system and it doesn't seem to be a fussy as Win11 is.  You aren't the first to have that sort of problem with the Cura installation.
It's good you got it sorted.