Ultimaker/Cura #15971: Support Raft Extends Off The B uild Plate

Kethiannne, 2023-07-01:
### Application Version

5.3.1

### Operating System

Windows 10

### Printer

Custom FFF Printer: Anycubic Kobra Neo

### Reproduction steps

A while ago I submitted this same issue with an older version of cura (4.12.0) and was told I should update and that it would fix the problem, my issue was then closed. I continued my project, printing out all the other pieces first so I could revisit this with a bit more experience, but even after updating cura and messing with my settings, this keeps happening with this piece.
I set my settings up like this:
[image]
[image]
[image]

### Actual results

the raft extends off the build plate
[image]

### Expected results

the raft wouldn't extend off the build plate

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[frame-holder.zip](https://github.com/Ultimaker/Cura/files/11926552/frame-holder.zip)

GregValiant, 2023-07-01:
Thanks for the report.
This is from your project file.
[image]

With the "Raft Extra Margin" at 9mm and the Support Brim enabled, it sticks off the bed.  If I increase your 9mm margin to 10mm it won't slice.
This is with the "Raft Extra Margin" at 6mm.  Everything fits.
[image]

This one is with "Raft Extra Margin" at 9mm but with "Enable Support Brim" turned off.  The supports are already on the raft.  Once again, everything fits.
[image]

So this appears to be happening because Cura is putting the supports on the raft and not counting the extra distance of the support brim.  It's also a matter of your settings.

I'll leave the bug label on this and the Cura team will take a look.

Kethiannne, 2023-07-01:
thank you! 
I was able to get it to fit by lowering the brim distance as well.
Enabling support brim being the issue makes sense, I didn't even think about that for some reason. Glad to keep learning!