Ultimaker/Cura #15972: Slicing failed as soon as the support structure is activated

Europeee, 2023-07-01:
### Cura Version

5.4.0 beta 1

### Operating System

Win 11 Pro

### Printer

Qidi I mate

### Describe your model

Dino Toy

### Add your .zip here ⬇️

The failure comes often, if I activate the "support structure". Without there is no problem (dosent matter the size of the object).
[image]
[image]

nbourre, 2023-07-02:
Same kind of error and might be related. Cura beta version fails to slice or refuse to when I use Customs supports cylinder.

[support test.zip](https://github.com/Ultimaker/Cura/files/11930025/support.test.zip)

I uploaded the project file.

MariMakes, 2023-07-05:
Hey @Europeee,

Welcome to the UltiMaker Cura Github 🚀 
Thanks for your report 👍

I believe what you are seeing is quite a common crash that we are trying to resolve. 
It sometimes goes away if you move the model away from the center of the buildplate?

Do you have a project file for us? It contains the printer and settings we need for troubleshooting. 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser. 

@nbourre, I believe what you are seeing is https://github.com/Ultimaker/Cura/issues/15836, where using custom supports would result in a slicing crash.  It's resolved in the 5.4 stable release, you can download the Cura version with the fix here: https://github.com/Ultimaker/Cura/releases/tag/5.4.0

Europeee, 2023-07-05:
Hi,short answer to your questions:It sometimes goes away if you move the model away from the center of the buildplate?Yes, that works from time to time. Do you have a project file for us? It dosent matter which object/model you want to print. It makes no difference.Best regards

MariMakes, 2023-07-11:
@Europeee,

Thanks for your update👍
The issue that I'm referring to should only appear on some models.
[Even with this small cube? ](https://www.thingiverse.com/thing:1278865)
If it appears on all your models then something is really wrong 😨 

Maybe it's a combination of settings in your profile. 
For that, I'll still need a project file to troubleshoot. 
Could you make a project file for me with a small group so I can take a look at the settings you use? 

To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the view it on GitHub link at the bottom of your email to open GitHub in the browser.

closed 2023-07-26 by github-actions[bot]