Ultimaker/Cura #15975: catastrophic bug

fabry2012, 2023-07-01:
### Application Version

5.3.1

### Operating System

Windows 11

### Printer

Anycubic Kobra MAx

### Reproduction steps

1. Rectangle raised 10mm on the z-axis
2. Use of adaptive layers activated
3. Slice

On the first layer there will be the following gcode with negative Z axis height which will make the nozzle hit the print bed violently, destroying the head and the bed. A real example

;LAYER:0
M107
M204 S20000
M205 X10 Y10
G1 F1200Z-24.29
;MESH:crown.stl
G0 F2700 X204.921 Y203.733 Z-24.29 <============

### Actual results

physical destruction of the nozzle and print bed

### Expected results

Z axis of the first layer close to zero or with a positive value

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

Configuration
[image]
bugged gcode
[image]

GregValiant, 2023-07-02:
Thanks for the report.
Please load the model and set Cura up to slice.  Then use the "File | Save Project" command to create a 3mf file.  Right click on the 3mf file in Explorer and select "Send To / Compressed (Zipped) Folder".  Post the Zip folder here.
I have never seen this behavior and this is the first bug report I have seen that describes this.
Do you have the "Z Offset Setting" or "Cura Slow Z" plugins loaded?

MariMakes, 2023-10-19:
Quick update from our side 👋

I stumbled upon your issue and recall bringing it up to the team but cannot find the follow-up we've done on it.
So I'll bring it up to the team again to see what they can do to improve it. 

Apologies for the slow response.

MariMakes, 2024-01-11:
Apologies for the delayed response. 

We've added a ticket to the backlog with the intent to improve this. 
For internal reference CURA-11537

Thanks for the report! 👍

HellAholic, 2024-03-01:
Issue has been resolved, validating with the project file @GregValiant provided (Thanks), the gcode shows: 
`G0 F9000 X98.5 Y103.121 Z0.2`
after applying the fix.

Fix will be part of the next release.

Cheers!

closed 2024-03-01 by HellAholic