Ultimaker/Cura #15976: Cant change extruder number

Phil1988, 2023-07-02:
### Application Version

5.3.1

### Operating System

Windows 19 x64

### Printer

Any

### Reproduction steps

1. Go to "Manage printers"
2. Open "Machine Settings"

### Actual results

You can only select one Extruder. In the past (at least I remember in Cura 4.x) it was possible to enter the number of extruders you want.

This is not possible any more.

Tested on mutiple printers. 
Its only possible to set the number as the default maximum that this printer was shipped.
Only via a drop down.

This breaks any possiblillity for multimaterial extrusion.

### Expected results

Let us pick as many extruder we want/have.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Here is the limiting drop down:

[image]

GregValiant, 2023-07-02:
Thanks for the report.
The printer must have multiple "Extruder Trains" defined in the main definition file and must have multiple "Extruder Definition Files" in order to be capable of having multiple extruders (up to 16 as I recall).  That is the expected behavior.
If you have customized the printer and added an extruder (or two) then you can either modify the printer definition file by adding additional extruder trains, and create additional extruder files, or you can add a Custom FFF printer and define additional extruders for it.
So this isn't a bug, it's the way it works and has for quite a while.  I have an Ender 3 Pro configured with 4 extruders and that's the way I had to do it.
I'll remove the bug label but I'll leave this open so you can respond.  I hope you understand.

Phil1988, 2023-07-03:
Hmm.... so this is indeed not a bug, but I would count this as a step back :D 
What exactly do I have to do to make it adjustable?

In the past I used Cura also for multi color prints of labels etc. by changing the filament manually.
This is now not possible any more. Very unfortunate.
Is there a how to on adding more extruders or maybe on multi color prints in this setup?

I have seen quite a few tutorials (written and from youtubers like TechingTech) that still explains the steps in a way that doesnt work any more.

Btw.: Thanks @GregValiant for responding so fast!

GregValiant, 2023-07-03:
I use Pause At Height.  The printer gets to a certain layer, retracts and raises, and then parks in a corner.  I change filament and click on the button on the LCD and the print resumes.  I should think that will work on your Voron.  The firmware must support M0 or M25 for the pause.
These are from my Ender with a single extruder.  The windows and curtains on the silly trailer are glued on but every other color change was done with Pause at Height.  I think there were 5 color changes on the coaster.
[image]

Phil1988, 2023-07-05:
Thanks @GregValiant for your tutorial, but your way only works with isolated colors at different layer heights.
I want to have real multiple color print.

Here is what I mean:
[image]

How can I archive this with Cura?

GregValiant, 2023-07-05:
It can be done.  It isn't fun and it isn't easy.
I call it "Emulate a Multi-Extruder Printer" and I've written a post-post processor (a separate app) to do it.  You have to configure a printer as multi-extruder and slice a multi-part file.  The app goes through the file and replaces all the tool changes with Pauses and you do a filament change at every pause.
For the model you show in the image there would be two color changes per layer.

In this two color benchy there are 173 pauses and filament changes in the print.  PITA.
If you want to play with the app I wrote (Windows only) let me know and I'll post a link to "Greg's Post Processors" here.
All those filament changes get real old real fast.
[image]

Phil1988, 2023-07-05:
Thats nice!

I would like to try that!
But doesnt that loop back to my original question, if I have to "configure a printer as multi-extruder" ? :D

GregValiant, 2023-07-05:
Yes it does.  You can install a "Custom FFF" printer and configure it however you want.  If you take the information from the Cura Machine Settings for your printer, and enter it into the Machine settings of the new printer, you should be good.  So long as the Firmware Flavor, bed size, Startup and Ending Gcodes are the same the "virtual" multi-extruder gcode will be printable on your printer.

There is a requirement in my app that the multi-extruder printer has to be configured with "Extruders Share Heater" and "Extruders Share Nozzle".  To access those settings you need to load the "Printer Settings" plugin from the MarketPlace.
This is a work in progress so consider it a beta version.  Let me know if there are issues.
[Greg's Post Processors](https://drive.google.com/uc?export=download&id=1PdNk-cPOW_VigyBDq8a_kcIlBxOd7Q_9)

Phil1988, 2023-07-06:
Ahh its a complete standalone post processing app.
Very nice!

I only print 2-3 times a year with this multicolor print and have done the gcode modifications by hand.
I will keep this and use it the next time. Thanks for sharing.
By the way: there might be others also interessted in this.
Any thoughts of publishing it on your git or similar?

GregValiant, 2023-07-06:
I haven't put it up yet.  I keep playing with it and it needs more debugging.  I don't want a bunch of people yelling at me.
I'll put up the install files later this month.  I think the PrusaSlicer capability is a plus as Prusa doesn't have post-processors.

Cromorg75, 2025-06-14:
hi , you can add extruder definition file in this pathway 
C:\Program Files\UltiMaker Cura 5.10.0\share\cura\resources\extruders

[image]

[image]

and you must change the voron2_base definition file in resources/definitions

[image]

Phil1988, 2025-06-15:
Yeah... such a workaround will work but something for the GUI would be the modern way of how to support something like this (in my opinion).

For the time, the workflow with GregValiants app worked fine.
But I have to admit that no native support is a downer for the usability and I think we are all very aware that multicolor printing is not a future feature but already all over the place.

Thats one reason why I dont use Cura any more on a regular basis.

Cromorg75, 2025-06-15:
i hope it will change in new versions