Ultimaker/Cura #15982: machine tooling

Kevinwall5011, 2023-07-02:
### Cura Version

5.3.1

### Operating System

windows 10 

### Printer

zonestar mk3z9 

### Name abnormal settings

no just did update to 5.4 have printed 7 times already before update 

### Describe model location

center

### Describe your model

tooling for machine 

### Add your .zip here ⬇️

[ZZ_MD TOOLING.zip](https://github.com/Ultimaker/Cura/files/11931675/ZZ_MD.TOOLING.zip)

GregValiant, 2023-07-03:
Thanks for the report and the file.
Two of the objects that make up the model are not "watertight".  In addition, there are some odd patches around the periphery like this rough area.  Is it supposed to be like that?
[image]

This area is very poor and probably won't slice correctly.
[image]

I'm going to remove the bug label from this and close it.  That model really needs repair and it will probably need to be done in CAD.  I wasn't able to repair the model using my regular utilities. 
If you repair the model and it still won't slice then we can re-open this. I hope you understand.

GregValiant, 2023-07-03:
I played with this some more and it does slice in 4.13.1.  It's still a bad model though.

Kevinwall5011, 2023-07-03:
I've printed the model 6 time now and printing one currently on different
software and it's rough because it's a subtract of the department of
defense helmets  and was generated in solid works also i was able to print
it on every  version but the 5.4.0 update and its a mold there's nothing to
change just gonna have to use older versions or software then  if you not
wanting to fix also have same slicing issue on 10 other models as well
thanks for the effort

GregValiant, 2023-07-03:
I'll re-open this then.  I sliced the model with PrusaSlicer and it doesn't look so good there either.   PrusaSlicer tells me there are 4,767,996 open edges.