Ultimaker/Cura #15983: Adopt "Save Project" / "Save Project As..." distinction for saving projects.

adrianlopezroche, 2023-07-03:
### Is your feature request related to a problem?

Saving projects in Cura currently involves the following steps: choosing "Save Project..." from the File menu, navigating to the desired destination folder, and giving the project a name. This is as expected. Saving *changes* to an existing project, however, requires an inordinate number of steps: selecting "Save Project..." from the File menu, navigating to the destination folder, choosing the current project's `.3mf` file, selecting "replace" when prompted by the operating system, and then again selecting "overwrite" when prompted by CURA. Additionally, overwriting the current project in this manner will also ask the user if they would like to reload the project, which is unnecessary because it's the same project.

### Describe the solution you'd like

Adopt a "Save" / "Save As..." distinction for saving files, the latter to be used when saving new projects (or saving existing projects under a new name), and the former when saving changes to an existing project. Saving changes should be a simple matter of choosing "Save," with no prompts of any kind. This is standard behavior for most programs.

### Describe alternatives you've considered

The standard approach is the best approach.

### Affected users and/or printers

This affects all users.

astillich, 2023-11-14:
This has driven me crazy far too many times. Would an implementation and patch be accepted by the Cura maintainers?

sausix, 2025-01-04:
Such a basic feature still missing.

Duplicate of #13264