Ultimaker/Cura #15985: Ender 3 S1 Pro Initial GCODE Bugs and Issues

chidiak, 2023-07-03:
### Application Version

5.4.0 Beta1

### Operating System

Windows 10 64x

### Printer

Ender 3 S1 Pro

### Reproduction steps

1. When making the first print, using the GCODE made by CURA, everything goes perfectly.
2. When performing the second print, it may be the same GCODE, if I move the bed manually, I have problems.
3. Apparently the printer does not get the initial positions of the axes.

### Actual results

Problems and crashes on the X axis.

### Expected results

Get the initial parameters of the axes at each print.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

**Gcode inicial cura 5.4**

; Ender 3 S1 Pro Start G-code
; M413 S0 ; Disable power loss recovery
G92 E0 ; Reset Extruder

; Prep surfaces before auto home for better accuracy
M140 S{material_bed_temperature_layer_0}
M104 S{material_print_temperature_layer_0}

G28 O ; Home all untrusted axes
G1 Z10.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X0 Y0

M190 S{material_bed_temperature_layer_0}
M109 S{material_print_temperature_layer_0}

G1 X0.1 Y20 Z0.3 F5000.0 ; Move to start position
G1 X0.1 Y200.0 Z0.3 F1500.0 E15 ; Draw the first line
G1 X0.4 Y200.0 Z0.3 F5000.0 ; Move to side a little
G1 X0.4 Y20 Z0.3 F1500.0 E30 ; Draw the second line
G92 E0 ; Reset Extruder
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X5 Y20 Z0.3 F5000.0 ; Move over to prevent blob squish

Asterchades, 2023-07-03:
I would suspect this is the result of using "G28 O" as opposed to simply "G28" to home the printer. Unless the steppers become disabled at the end of the previous print, or simply time out (subsequently disabling them), the printer will simply assume that all axes are accurate. Being that most printers use open loop positioning systems there is no way for it to know if you've manually moved anything, so the only assumption it can make is that you haven't while the steppers are still engaged.

For your local installation changing the above command will force it to re-home every time, and honestly is probably a good practice anyway. I would also recommend not moving the bed, carriage, or gantry by hand to avoid complications.

Note that all non-Ultimaker profiles are submitted by the community - Ultimaker and the Cura team try to remain as hands-off as possible for these. This particular change, for example, was made by [this pull request](https://github.com/Ultimaker/Cura/commit/a6e89a9d544029d99488fc0752a3061c5710f3ae) by @sparkym3 on January 20. Should you want to update the base profile for all installs I would encourage you to make a pull request with an appropriate change. The file in question is [here](https://github.com/Ultimaker/Cura/blob/main/resources/definitions/creality_ender3s1pro.def.json), with line 27 containing the start GCode.

closed 2023-07-03 by Asterchades

chidiak, 2023-07-06:
@Asterchades Thank you very much! Replaces the "G28 O" command with "G28", now everything is working OK!