Ultimaker/Cura #15988: Inconsistent speed calculations

vinridge, 2023-07-03:
### Application Version

5.3.0

### Operating System

Windows 10

### Printer

FLSUN V400

### Reproduction steps

I have been using Cura 5.2.2 for a while now with no issues. When i downloaded 5.3.1 and now 5.4.0, theses newer versions do not calculate speeds like any of the versions prior to 5.3.1. In 5.2.2 an previous versions, when you set the Print Speed, Cura will automatically scale all of the other print speeds (wall speed, travel speed, first layer speed, etc). In 5.3.1 and above the scaling seems way off and nothing like the previous versions where the speed scaling made sense. I have attached some screenshots of the scaling difference. In 5.3.1 and 5.2.2. Cura 5.3.1 is scaling many other speeds way higher than the set Print Speed. And there are "calculate" buttons ( ƒx ) but when you click these the scaling is still pretty far off from previous versions.

### Actual results

I would like to use the newer versions for the added features but need to figure out why the speed scaling if so off in newer versions.

### Expected results

Let me know if I need to supply any other data.

Thanks,

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]
[image]

GregValiant, 2023-07-03:
Thanks for the report.
The base file for your printer is fdmprinter.def.json.  Within that file some of the speeds have formulas and that's what is used when they are linked to Print Speed.
"speed_wall"
"value": "speed_print / 2"
and
"speed_infill"
"maximum_value": "math.sqrt(machine_max_feedrate_x ** 2 + machine_max_feedrate_y ** 2)"
I think one problem here is that the "circle" icons that should replace the fx icons aren't visible.  I think they are black and can't be seen when you are in "dark mode".

vinridge, 2023-07-03:
So I took it out of Dark Mode and the circles are not there. I know what you are talking about, they show up blue in Dark Mode.

[image]