Ultimaker/Cura #15989: Cura Freeze and/or crash apond moving object

Carlosloureiro4, 2023-07-03:
### Application Version

5.3.1

### Operating System

windowns11

### Printer

ender pro 3

### Reproduction steps

Apond moving a object cura feezes and if it comes back does it again its unusable. I tried uninstaling and reinstalling to no availe.

### Actual results

freeze or crash

### Expected results

freeze or crash

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

How?

Carlosloureiro4, 2023-07-03:
I tried contaction the support team but i didnt get a responce (19/06/2023 - 03/07/2023)

Carlosloureiro4, 2023-08-12:
> Hey @Carlosloureiro4 <https://github.com/Carlosloureiro4>,
>
> Welcome to the UltiMaker Cura Github 🚀
> Sorry, it took us a while to get back to you 😞
>
> The new Cura 5.4 seems to fix a number of slowness and crashing issues for
> people with Windows 11 Machines.
> Could you test if that's the case for you as well?
> You can download it here:
> https://github.com/Ultimaker/Cura/releases/tag/5.4.0
>
> If you still run into a crash, please double-check if you shared the
> correct file and let us know.
> I'll take another look at what is going wrong. 👍

Hello Mariska from Ultimaker/Cura yes the update from what I have tested so
far as solved the crash issues and the extreme slowness however for sizable
files it still takes time to move the objects but it's workable.

Thanks for the response and the help,
Carlos Loureiro.

MariMakes, 2023-08-14:
Hey @Carlosloureiro4,

Thanks for the update.
In that case, I'll be closing this issue. 

Thanks again, and please let us know if you run into any other issues 💪