Ultimaker/Cura #15991: I have been trying to slice a part but it wont

HIDUDEWAS, 2023-07-03:
### Application Version

5.3.1

### Operating System

windows11

### Printer

ender3v2

### Reproduction steps

I put in my model and clicked slice and it said that there was a bug

### Actual results

It say there is a bug

### Expected results

it slices and i can print it

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

it says slicing  failed with an unexpected error

GregValiant, 2023-07-03:
Thanks for the report.
We need a project file to investigate the issue.  Use "File | Save Project" then zip the 3mf file and post the zip folder here.

MariMakes, 2023-07-26:
Hey @HIDUDEWAS,

Have you tried reverting to the default settings?
Have you tried slicing with Support disabled? 
Have you tried moving the model away from the center of the buildplate and rotating by 90 degrees?
Can you check if your mesh is faulty with the Meshtools Plugin? You can get it here: https://marketplace.ultimaker.com/app/cura/plugins/fieldofview/MeshTools

If that's not it, I'll need a little bit more information from you.
Do you have a project file for us? It contains the printer and settings we need for troubleshooting. 
To save a project file go to File -> Save project. Please make sure to .zip your project file. For big files, you may need to use WeTransfer or similar file-sharing sites.

You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser. 

Also, the new Cura 5.4 contains the fix for this for a number of slicing crashes. 
I was able to slice your model successfully in 5.4 Stable.
Could you test if that's the case for you as well? 
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.4.0

If you still run into a crash, please double-check if you shared the correct file and let us know.
I'll take another look at what is going wrong. 👍

closed 2023-08-10 by github-actions[bot]