Ultimaker/Cura #15996: Cura does not report its installed size within the Apps control panel on Windows 10 or 11

meyersjd, 2023-07-04:
### Application Version

5.21 and 5.31

### Operating System

Windows 11 and Windows 10

### Printer

CR6SE

### Reproduction steps

1. Install the program
2. Look in the Apps control panel, formerly called Add/Remove Programs
3. The installed size is not listed next to either version.
[image]

### Actual results

No installed size is shown.
[image]

### Expected results

Apps panel should show installed size.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

n/a

GregValiant, 2023-07-04:
I don't see it in Win10.  The Cura installation folder and the configuration folder add up to about 1.9gb.
I have MS Visual Studio installed and it doesn't report the installation size either.

jellespijker, 2023-07-05:
dis you use the exe installer or the msi? I believe it is a known issue for the exe installer but the msi should report it correctly

meyersjd, 2023-07-05:
It was an exe installer. That is the one on the ultimaker website

MariMakes, 2023-07-26:
Hey @meyersjd,

To me, it doesn't seem like a bug in Ultimaker Cura, since we have a solution in place with the MSI. 
Apologies for any inconvenience caused by our delayed response. 
Would you still like to keep this issue open or do you agree that we can close it?

MariMakes, 2023-07-26:
Oh, in case you missed it, you can download the MSI here: https://github.com/Ultimaker/Cura/releases/tag/5.4.0

meyersjd, 2023-07-26:
Yes, the MSI works. Thanks! You can close it.

closed 2023-07-26 by meyersjd