Ultimaker/Cura #16000: Error when opening Cura 5.4.0 - ModuleNotFoundError: No module named 'PyQt6.QtNetwork'

mzac, 2023-07-04:
### Application Version

5.4.0

### Operating System

Windows 10

### Printer

Ender 3 Pro

### Reproduction steps

1. Installed Cura 5.4.0
2. Got error
3. Removed Cura 5.4.0
4. Re-downloaded the installer
5. Tried again, got same error

### Actual results

Traceback (most recent call last):
  File "cura_app.py", line 23, in <module>
ModuleNotFoundError: No module named 'PyQt6.QtNetwork'

[image]

### Expected results

Should not get an error

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

My old Cura 5.3.1 on this same computer still works.

MariMakes, 2023-07-04:
Hi @mzac,

Thanks for your report 👍
That does not look good 😨 
I'm not getting this error, but it looks serious.

Would you mind sharing your log files. You can find your log file here:
Windows: %APPDATA%\cura\5.4\cura.log

I raised it with the team, hopefully we can prevent more people from running into this error. 
Fingers crossed 🤞

mzac, 2023-07-04:
> Hi @mzac,
> 
> Thanks for your report 👍 That does not look good 😨 I'm not getting this error, maybe because I've already had Cura 5.4 beta installed, but it looks serious.
> 
> Would you mind sharing your log files. You can find your log file here: Windows: %APPDATA%\cura\5.4\cura.log
> 
> I raised it with the team, hopefully we can prevent more people from running into this error. Fingers crossed 🤞

Hm, this sucks, it didn't create a log file and nothing about 5.4 in the stderr.log  Anything else I can check?

[image]

Hedan-13, 2023-07-05:
Hi!!! GM
I'm having the same problem. Just about 10 minutes download from the official site of Ultimaker, install Cura 5.4.0 and get this error. My Cura 5.3.1 still works. 
Also, didn't create any log file for 5.4
TX!!!
[image]

janderek, 2023-07-10:
Getting the same fail, just tried to install at a win10 machine, and then it dosnt fail, but on my win11 machine it wont work, 

as the others, the 5.3.1 i still working

leftoverjacksons, 2023-07-11:
Having the exact same issue, windows 11

Nasersanji, 2023-07-21:
I have exact the same! :(

MariMakes, 2023-07-25:
Hey all,

We discussed this issue with the team.
We are a bit at a loss what is happening here 😖 

Can you try the following things?
- Can you try to install the MSI instead of the .EXE? You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.4.0
- Can you try to run the program as an administrator?  
[image]

Please let us know if these options do or do not work for you, then we'll try to have another look.

Hedan-13, 2023-07-25:
Good news!!! After unistalling the version 5.3.1, and deleting all the files related to older versions. Run the MSI as administrator and now is working.  
[image]

mzac, 2023-07-26:
I also tried the MSI vs the EXE and it worked.  Didn't have to remove 5.3.1 or run as administrator, it just works.  Thanks for looking into this but it would still be nice to understand what went wrong here.

MariMakes, 2023-07-27:
Hey all,

Thanks for reporting back!
I'm glad we found a workaround. 😊 

I would also like to understand what was going wrong here. I believe it's related to the updated PyQT library we introduced to fix a host of issues. You can read more about it here: https://github.com/Ultimaker/Cura/discussions/15098

To me, it doesn't seem like a bug in Ultimaker Cura, since we have a solution in place. 
I'll be closing this issue. I hope you understand.

closed 2023-07-27 by MariMakes