Ultimaker/Cura #16002: Failed to slice error

dcoleman4, 2023-07-04:
### Cura Version

5.4.0

### Operating System

Windows 10 x64

### Printer

Creality Ender 3 V2 NEO

### Describe your model

Model is watertight. Has been successfully sliced on Cura 5.3.1. All Meshtool plugins has been ran. Object scaled to 125%

### Add your .zip here ⬇️

[CE3E3V2_3topofshelf.zip](https://github.com/Ultimaker/Cura/files/11951529/CE3E3V2_3topofshelf.zip)

HellAholic, 2023-07-05:
Hey @dcoleman4,
I did some investigation on the project file you shared and found the cause for your troubles.
Since the model is close to your build-plate dimensions, and the skirt distance has a default value of 10.0mm, it causes a fail response, even though you do not have the build-plate adhesion enabled.
This is indeed an Edge-case, but it does increase the value of your report.

[image]

I do have a workaround for you.

1. Go to Preferences
2. Configure Cura
3. Settings
4. Enable Skirt Distance
5. Change the Skirt Distance to 4.5mm
6. Keep the build-plate adhesion disabled

Can you let us know if that works for you?

We've added a ticket to the backlog with the intent to improve this. 
For internal reference - CURA-10758

Thanks for the report! 👍

dcoleman4, 2023-07-11:
Hey @HellAholic! 

Thank you for helping me out! That did fix the issue I had. Thanks for your help!

closed 2023-07-11 by dcoleman4