Ultimaker/Cura #16003: Can't load multiple files at the same time

xeddog, 2023-07-04:
### Application Version

5.4.0

### Operating System

Windows 11 Home

### Printer

Custom

### Reproduction steps

Cura 5.4  won't allow loading multiple files at the same time.  For example,
- click on the load icon.
- navigate to the folder with stl files.
- select first file
- Move cursor to last file
- shift-click last file
the first item is DE-selected and the last file, and only the last file, is selected

The same thing happens with ctl-click

Computer is a Windows 11 laptop, and this process works with all other applications, etc. 

### Actual results

Only one file is loaded.

### Expected results

All in the selected range for shift-click, or individual selected files in the case of CTL-Click, should be loaded

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

No additional info

GregValiant, 2023-07-04:
Thanks for the report.
This is a duplicate of #15879 and the behavior seems to have appeared in 5.4beta.  Cura 5.3.1 allowed the opening of multiple files at the same time.
I can confirm on my system that I can only open one at a time.  So far the other report and my own system are on Windows.
Maybe @rburema of the Cura Team has some information?

xeddog, 2023-07-04:
I found a workaround.  It is still a bit of a nuisance, but hey.  Just have to go to the file manager first, select the files I want, and then drag them to the Cura window and drop them on the bed.

xeddog, 2023-07-05:
Thanks for the reply.  I thought I reported this a couple of weeks ago, but I didn't realize at the time that I was on the forum, and not here.  I hate getting old..   :-)

Wayne

rburema, 2023-07-05:
I can confirm that this doesn't work anymore, and did in 5.3.x

I don't have any other info other than that at the moment, but I'll put it on the board to discuss.

MariMakes, 2023-07-10:
This seems to be a duplicate of https://github.com/Ultimaker/Cura/issues/15879
We have a ticket on our backlog with the intent to improve this behavior.
I'll be closing this issue as a duplicate but you can follow the progress here: https://github.com/Ultimaker/Cura/issues/15879

Thanks for your report 👍

closed 2023-07-10 by MariMakes