Ultimaker/Cura #16007: Elegoo Neptune 3 Pro NO Profiles

jzappitello, 2023-07-04:
### Application Version

5.4.0

### Operating System

MacOS Ventura

### Printer

Elegoo Neptune 3 Pro

### Reproduction steps

1. I just tried selecting my printer and I even tried changing the material type fro Generic PLA but it always gives me the message, “No profiles are available for the selected material/Nozzle Size configuration. Please change your configuration."
2. I attempted to copy the same GCode and machine settings from the elegoo cura version but that did not help either.
3. I did notice that inside the “Machine Settings” for the printer and on the “Extruder 1” tab there is no “Nozzle Settings” to change the nozzle size like in the elegoo cura version. I think this might be related?

### Actual results

I am still met witht he same error -> “No profiles are available for the selected material/Nozzle Size configuration. Please change your configuration."

### Expected results

I would have expected to be able to see this error go away and it resolve my issue.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]
[image]
[image]
[image]

NoReMac2, 2023-07-04:
Same issue for the Neptune 3. Cura told me that I had some corrupted profiles that I needed to remove, but then there were no profiles for my printer.

commander55, 2023-07-05:
Same issue, full Cura reset did not change it.
[image]

TheNvisible, 2023-07-06:
Add me to the list of those having the trouble.

Marciemousepad, 2023-07-07:
I'm Having the same Issue Im on OS Monterey and even reinstalling 5.3 it has corrupted my profiles there too

ShaalanAhmed, 2023-07-08:
I have the same issue with Cura 5.4.0 with Neptune 3 Pro as well. I had to fall back to 5.3.1

sunny852206, 2023-07-12:
Same issue for Neptune 3 Pro with Cura 5.4.0. Been told had corrupted profiles and needed to reset them.

ShaalanAhmed, 2023-07-12:
> Same issue for Neptune 3 Pro with Cura 5.4.0. Been told had corrupted profiles and needed to reset them.

did it work? and how did you reset it please?

strenkml, 2023-07-19:
I am also having the same issue with my Neptune 3 Plus

bmyhny, 2023-07-22:
same issue

AniMeiGrrl, 2023-07-25:
I also have the same issue. I've had the 5.3 profiles corrupted too and even a full reinstall, cleansing of the Application Support folder, none of it resolves the issue. I have to set up all of my Elegoo printers as generics and frankly this sucks :D

MariMakes, 2023-07-26:
Hey All👋

I think what you are seeing might be related to an update in the printers. You can read more about it here: https://github.com/Ultimaker/Cura/pull/14946
We also had a bug in the beta, that materials were missing, but that should be resolved in the 5.4 Stable. 

I can help better with troubleshooting if I could have a look at your configuration folder. 
You can share your configuration folder by:
	1. Go to Help at the top bar of Cura
	2. Select Show Configuration Folder
	3. Select all the folders in there and create a zip OR go up a folder in your folder structure and zip your folder with the correct Cura version
You'll need to open Github in your browser to attach any files to your ticket
You can click the **view it on GitHub** link at the bottom of your email to open GitHub in the browser.

AniMeiGrrl, 2023-07-26:
Thank you for your reply! Here is my entire 5.4 folder - right now I'm just getting by by using custom profiles but I really thank you for looking at this! :)

[5.4.zip](https://github.com/Ultimaker/Cura/files/12171838/5.4.zip)

commander55, 2023-07-27:
> Same issue, full Cura reset did not change it. [image] [Cura 5.4 Mac.zip](https://github.com/Ultimaker/Cura/files/12179663/Cura.5.4.Mac.zip)

Thanks for looking into it. Updated my post

MariMakes, 2023-07-27:
We believe it might be related to the upgrade script introduced in Cura 5.4 and it needs some more investigation 

@Joeydelarago your change seems to have introduced some breaking configurations.
I'll try to reproduce it by running an upgrade test, would you also mind taking a look at what is going wrong? 🙏

AniMeiGrrl, 2023-07-27:
If there is anything we can do from a user standpoint to troubleshoot or test, I am more than happy to help.

Joeydelarago, 2023-07-27:
> We believe it might be related to the upgrade script introduced in Cura 5.4 and it needs some more investigation
>
> @Joeydelarago your change seems to have introduced some breaking configurations. [...]

Oops, I'll have a look at it over the weekend and create a PR.

MariMakes, 2023-07-28:
Ok, quick update from my side.

I've tried to reproduce what you are seeing by doing a clean install and than doing the upgrade flow in Cura 5.2-5.3-5.4 to see if I could encounter it as well, but I wasn't able to. The materials are showing up correctly for me, which makes me think that it's related to how your Cura has been installed. 

I can see that in your logs it says: ` Could not find preferred quality type Elegoo_layer_020 for material generic_pla_175 and variant elegoo_neptune_3pro_0.40, falling back to empty_quality.`
So I do have a carepackage for you folks here to at least get unstuck. 😊 

On Windows navigate to to: C:\Program Files\UltiMaker Cura 5.4.0\share\cura\resources
On Mac, right-click on UltiMaker Cura.app in your applications folder and click "Show Package Contents"
navigate to ../Contents/Resources/share/cura/resources/

If you download this zip: [ElegooCarePackage.zip](https://github.com/Ultimaker/Cura/files/12194002/ElegooCarePackage.zip)
You can drag the content of these folders to your resources, you should have all the files you need again. 
[image]

I hope at least that helps you get unstuck 🤞

AniMeiGrrl, 2023-07-28:
That did indeed do it as a quick fix for me! 

I will say that every report I have seen of this in FB groups and such has been Mac related. Also, this does happen to me even on fresh installs - not just upgrades.

Hope that helps!

commander55, 2023-07-28:
Did not work for me (Mac). Even removing 5.4 config folders and a fresh installation did not fix it. Removed all 5.4 configs and now back on 5.3.1.

seehundnz, 2023-07-28:
Sorry, same here. The CarePackage did not work on my System with a clean installation on Mac (Intel)

z-drucker, 2023-07-29:
Unfortunately, I am having the same issue on an M1 Air.  The care package doesn't work... :(

jzappitello, 2023-07-31:
> Hey All👋
>
> I think what you are seeing might be related to an update in the printers. You can read more about it here: #14946 We also had a bug in the beta, that materials were missing, but that [...]

I have just edited my comment and attatched my config file. 
I want to also note that I did this after two things:
1.  A fresh install of 5.4.0 
2. I followed all of the instructions of the above mentioned “ElegooCarePackage”

Hopefully you can give some insight as to why I’m still having this issue. I am on an M1 chip MacMini -if that gives any insight into anything either.

MariMakes, 2023-08-02:
Hey all,

Thanks for the updates, that can help narrow down what is happening. 
For the Mac users this is another workaround that might work. 

1. [Open Terminal](https://www.howtogeek.com/682770/how-to-open-the-terminal-on-a-mac/)
2. Type or copy paste ` open ~/Library/Application\ Support/cura`
3. Open 5.3 folder
4. Navigate to ../Contents/Resources/share/cura/resources/
5. Drag the contents of these folders to your resources

aikiharp, 2023-08-03:
> 5. Drag the contents of these folders to your resources

@MariMakes  Thanks for your persistence in helping us get this working! Where you say "Drag the contents of these folders to your resources," do you mean we should copy _all_ the contents of ~/Library/Application Support/cura/5.3 to /Applications/Ultimaker Cura/Contents/Resources/share/cura/resources/ ? 

Here's what I've done so far:

1. I tried putting the contents of the care package from your post above in the application bundle and that did not overcome the extruder profile error (although Cura would launch albeit with Gate Keeper warnings), so I removed the app and its config folder. 
2. I installed a fresh copy of 5.4.0 and copied the same folders named in the care package from my 5.3 config folder to .../resources in the app bundle. The app didn't migrate anything, and asked me to set up my printer as new. Once I chose Elegoo Neptune 3 Pro and clicked Next, Cura crashed. Relaunching Cura resulted in the same behavior.
3. Then I copied _all_ the contents of the 5.3 config folder into the indicated location in the app bundle. This approach prevented the app from launching at all, and throwing the following error: 
[image]

I've been checking daily to see if there's a patch or updated release. Troubleshooting this is beyond my capability, but I sure would like to get 5.4.0 working.

z-drucker, 2023-08-03:
> Hey all,
>
> Thanks for the updates, that can help narrow down what is happening. For the Mac users this is another workaround that might work.
>
> 1. [Open [...]

Unfortunately, this doesn't work.  I just tried with fresh installs as well (I did clear the Application Support folders before fresh install, too).... boo...

Doormandean, 2023-08-05:
Yep, M1 Mac mini user here, tried everything above and no improvement. 

Bit rubbish really, having to print things in general and it's taking forever

dapetcu21, 2023-08-06:
Same issue here. 5.4.0 doesn't work. 5.3.1 works. M1 MacBook Air

aikiharp, 2023-08-08:
@MariMakes has this bug been resolved since the issue is closed? It looks like the workarounds aren't working for most people here.

MariMakes, 2023-08-09:
Hey @aikiharp, 

We are very much still looking into this issue. 👍 
We do have a tendency to close duplicates to keep an overview, perhaps that is what you are seeing. 

Something seems to go wrong during the Cura installation on Mac devices for certain printer definitions if I read the comments correctly. (If you are not running a MacOS please let us know)

One of our resident Mac Developers is checking the workflow.

sreekarchigurupati, 2023-08-16:
Any update? Just bought a Neptune 3 Pro and installed Cura and am having the same issue

janv37, 2023-08-16:
Hi, this is opensource project, so just give devs the time they need to address the issue. I'm running Ventura and have Neptune 3 Pro too. Cura 5.3.1 works just fine, so maybe use that as a workaround? Have a nice one!

sreekarchigurupati, 2023-08-16:
@janv37 just downgraded to 5.3.1 and it's working fine. I appreciate the developers efforts. Just wanted to share that I'm facing this issue too

rakeltner, 2023-08-16:
Same issue for me as well Mac M2

T3rm1, 2023-08-18:
It's been a problem since the beta and it has been known for nearly two months now with no solution.
There are other bugs also related to Neptune 3 Pro but no it seems that no one is working on them. The repositorty has over 2k open issues and most of them are ignored and don't even have a comment from a responsible person.

janv37, 2023-08-18:
I don't have an idea, if mac version of cura is done by somebody as a dayjob or it's somebody's hobby in the evenings. What I know, it's opensource. The code is here and it's open. Where is your pull request with the fix? 
Why do you think you're entitled to somebody else's time?

janv37, 2023-08-18:
Okay, this works for me:

0. remove pre-existing 5.4 resources in library (backup if you need it):
`rm -rf ~/Library/Application\ Support/cura/5.4`
2. download Ultimaker Cura 5.4
3. open dmg and copy Ultimaker Cura to Applications
4. right click and open Ultimaker Cura in Applications to open unsigned app (confirm open), skip wizard and close Cura. (This creates basic resources in Library (~/Library/Application\ Support/cura/5.4) 
5. Download  ElegooCarePackage.zip above
6. unzip ElegooCarePackage.zip in downloads:
`cd ~/Downloads && unzip  ElegooCarePackage.zip`
7. copy 4 directories from ElegooCarePackage to resources folder in Library. The cp command merges with what's already there. Finder's default behavior is to replace. (There might be empty directories, I haven't checked though)
`cd ~/Downloads/ElegooCarePackage && cp -r extruders intent quality variants ~/Library/Application\ Support/cura/5.4`
8. Re-run Ultimaker Cura from Applications
9. Add Elegoo printer - I have Elegoo Neptune 3 Pro, the exclamation point (!) on Generic PLA is gone and I finally have Profiles different from "Fast 0.2mm", I can import STL file and Slice it. Will print later.
10. Profit! 

Big thanks to DEVs for the workarounds if they can't roll new release just yet. Those two single steps didn't work for me before, so I adapted it a bit. Also, editing resources directly in /Application/Ultimaker Cura/resources probably messes up some checksum, so the app can't open. That's why I copied over to the ~/Library/Application\ Support/cura/5.4

P.S. This is on my Monterey Laptop, but seen same issue on Ventura too.

commander55, 2023-08-18:
> Okay, this works for me:
> 
> 0. remove pre-existing 5.4 resources in library (backup if you need it):
>    `rm -rf ~/Library/Application\ Support/cura/5.4`
> 1. download Ultimaker Cura 5.4
> 2. open dmg and copy Ultimaker Cura to Applications
> 3. right click and open Ultimaker Cura in Applications to open unsigned app (confirm open), skip wizard and close Cura. (This creates basic resources in Library (~/Library/Application\ Support/cura/5.4)
> 4. Download  ElegooCarePackage.zip above
> 5. unzip ElegooCarePackage.zip in downloads:
>    `cd ~/Downloads && unzip  ElegooCarePackage.zip`
> 6. copy 4 directories from ElegooCarePackage to resources folder in Library. The cp command merges with what's already there. Finder's default behavior is to replace. (There might be empty directories, I haven't checked though)
>    `cd ~/Downloads/ElegooCarePackage && cp -r extruders intent quality variants ~/Library/Application\ Support/cura/5.4`
> 7. Re-run Ultimaker Cura from Applications
> 8. Add Elegoo printer - I have Elegoo Neptune 3 Pro, the exclamation point (!) on Generic PLA is gone and I finally have Profiles different from "Fast 0.2mm", I can import STL file and Slice it. Will print later.
> 9. Profit!
> 
> Big thanks to DEVs for the workarounds if they can't roll new release just yet. Those two single steps didn't work for me before, so I adapted it a bit. Also, editing resources directly in /Application/Ultimaker Cura/resources probably messes up some checksum, so the app can't open. That's why I copied over to the ~/Library/Application\ Support/cura/5.4
> 
> P.S. This is on my Monterey Laptop, but seen same issue on Ventura too.

Thats it!! 3 folders have been empty, but one already contained some files. Merging those did the trick. Thanks a lot!!

Joeydelarago, 2023-08-19:
Hey everyone, I am trying to fix this issue but I can't reproducing this problem with the linked config folders. Could someone else with the issue upgrading 5.3 -> 5.4 share their 5.3 and 5.4 config folder with me? Is anyone having this issue on Linux/Windows or is it only MacOs?

Apoligies about the issues, the aim of the changes I made was to add much faster profiles (similar to the Ultimaker printers). As well as supporting standard size nozzles (0.2, 0.4, 0.6, 0.8). I think these profiles are a big improvement!

janv37, 2023-08-19:
Hello @Joeydelarago,
here are subfolders from ~/Library/Application Support/cura on MacOS Ventura, English locale (probably does not influence anything)

  * Both 5.3 and 5.4 working fine, 5.4 has the carepack applied the way I wrote 2 posts above
[5.3.tar.gz](https://synchronicity.cz/cura/5.3.tar.gz)
[5.4-withcarepack.tar.gz](https://synchronicity.cz/cura/5.4-withcarepack.tar.gz)

I've deleted both 5.3 and 5.4 directories ( at ~/Library/Application Support/cura ) at this point
  * This gets created on clear start of 5.4 and skipping intro and adding Neptune 3 Pro as non-networked, non-ultimaker printer. There's exclamation mark at the Generic PLA and 0.40mm_Elegoo_Nozzle in the middle of the screen and only Fast profile on the right hand side. Details say Configuration not supported. (screenshot below)
[5.4-clearstart.no53.tar.gz](https://synchronicity.cz/cura/5.4-clearstart.no53.tar.gz)

  * This is 5.4 start with existing 5.3 directory around (migration (?))
[5.4startwith5.3profile_in_place.tar.gz](https://synchronicity.cz/cura/5.4startwith5.3profile_in_place.tar.gz)

[image]

Ping me if you need a video, or more screenshots or something. Also, if you get 421 error on accessing those .tar.gz files, I've messed up config of this virtualhost and GeoIP filter is still in place. Let me know in that case, please, too. (There should be none)

janv37, 2023-08-19:
FWIW, my brother uses 5.4 with Elegoo Neptune 3 Pro on Windows 10, czech locale and all the Elegoo settings are there and work just fine.

janv37, 2023-08-19:
And I've been "installing" Cura 5.3 and 5.3 on Monterey and Ventura from the dmg file. I've seen there's pkg file available too - so maybe you used that and that's why you can't reproduce the problem?
Could you describe your setup, so we could compare and hunt this down?

janv37, 2023-08-19:
Ah, okay, I can upload here too. 

[5.3.tar.gz](https://github.com/Ultimaker/Cura/files/12386401/5.3.tar.gz)
[5.4-clearstart.no53.tar.gz](https://github.com/Ultimaker/Cura/files/12386402/5.4-clearstart.no53.tar.gz)
[5.4-withcarepack.tar.gz](https://github.com/Ultimaker/Cura/files/12386403/5.4-withcarepack.tar.gz)
[5.4startwith5.3profile_in_place.tar.gz](https://github.com/Ultimaker/Cura/files/12386404/5.4startwith5.3profile_in_place.tar.gz)

janv37, 2023-09-02:
So I've just installed Ubuntu 22.04 LTS on kid's laptops and Cura from Appimage for modern linux has all the Elegoo profiles that are necessary. So this issue really seems to be MacOS-only.
JV

tejamainhoon14, 2023-09-04:
same issue with mac M1 max, ventura 13.5.1 and cura 5.4

saumyaj3, 2023-10-05:
Hi! This issue has been fixed for the next upcoming Cura release 🗡️

MariMakes, 2023-10-11:
This is resolved for Elegoo but also for  Strateo3D, Uni, and ZAV 3D printers in the 5.5 Beta release 🎉 , you can download the Cura version with the fix here: [https://github.com/Ultimaker/Cura/releases/tag/5.5.0-beta.1](https://github.com/Ultimaker/Cura/releases/tag/5.5.0-beta.1)
I'll close this issue since it's resolved. 

Thanks again, and please let us know if you run into any other issues 💪

PR #16784 (2023-09-22): Only filter files in directories starting with .
# Description

Mac builds were missing some definitions, which happened because there was
a . in the directory. The reason why we filtered these in the first place
was because Apple doesn't notarize/sign an App correctly if the directory
starts with a . e.q.: .dylibs in scipy and shapely.

Renamed the resources folder containing a `.` with an `_`

Added a new diagnostic check to the printer-linter which will diagnose if the directory contains a `.`

```
Diagnostics:
-   DiagnosticMessage:
        FileOffset: 1
        FilePath: ./ultimaker/Cura/ [... cut]