Ultimaker/Cura #16008: 5.4  Ender 3 S1 Pro Profile Bug

mdalamond, 2023-07-05:
### Application Version

5.4.0

### Operating System

Windows 11

### Printer

Creality S1 Pro

### Reproduction steps

I updated to 5.4.0 and loaded the Creality S1 Pro profile.

### Actual results

On Start-up of the print bed slammed into the back of the machine.  
Once I started printing The first 6 or 7 layers step in as the layer go up. After that it prints like it has layer shift and there are gaps between the layers. 

### Expected results

it should have printed like 5.3.0 and 5.3.1

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

I loaded a good custom profile into 5.4.0 that I was using on 5.3.0 and 5.3.1 and it prints fine. I am new to the 3d printing world, but this seems like a profile issue.

GregValiant, 2023-07-06:
Thanks for the report.
Go to "Settings/Printers/Manage Printers and then Machine Settings.
Your StartUp Gcode (as stated in the printer definition file) might have a line "G28 O ;home all untrusted axes".  That line might cause the problem because if the printer knows where the X and Y are but not the Z, then the Z will home wherever it happens to be and I can see that hitting the bed..  I would suggest you change it to a simple "G28".  That will cause the printer to home X and Y first putting the nozzle over the left front corner and then Z will home and it should not hit the bed.

You need to post a Project File here as well.  Load something simple like a calibration cube and use the "File | Save Project" command and then Zip the file and post the zip folder here.
The printer definition files come from either a member of the community or from Creality.  UltiMaker does not provide definition files for 3rd party printers and has no way to check to see if the submitted definition is correct.
Post that project file and I'll take a look.
I'll leave the bug label on for now but if the problem is the definition file then it really isn't an UltiMaker Cura bug and the definition will need to be changed.

mdalamond, 2023-07-06:
Hi Greg,
Thank you for your response. Like I said Im new at this so I will do my
best. I will post a project file as soon as I can.4
Here is the start up G-Code below. This is the one on the new 5.4.0 version.

; Ender 3 S1 Pro Start G-code

; M413 S0 ; Disable power loss recovery

G92 E0 ; Reset Extruder

; Prep surfaces before auto home for better accuracy

M140 S{material_bed_temperature_layer_0}

M104 S{material_print_temperature_layer_0}

G28 O ; Home all untrusted axes

G1 Z10.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed

G1 X0 Y0

M190 S{material_bed_temperature_layer_0}

M109 S{material_print_temperature_layer_0}

G1 X0.1 Y20 Z0.3 F5000.0 ; Move to start position

G1 X0.1 Y200.0 Z0.3 F1500.0 E15 ; Draw the first line

G1 X0.4 Y200.0 Z0.3 F5000.0 ; Move to side a little

G1 X0.4 Y20 Z0.3 F1500.0 E30 ; Draw the second line

G92 E0 ; Reset Extruder

G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed

G1 X5 Y20 Z0.3 F5000.0 ; Move over to prevent blob squish

 As you mentioned it appears to have the G28 O. I will give that a shot.
Im in the middle of some printer upgrades at the moment but will try it
when Im done.
Thanks again, Im learning new stuff at an old (er) age, lol
Matt

mdalamond, 2023-07-06:
[CE3S1PRO_female2c2.zip](https://github.com/Ultimaker/Cura/files/11972176/CE3S1PRO_female2c2.zip)

GregValiant, 2023-07-07:
I think that's it.  I'm not sure what conditions would have to happen for the Z to be the only stepper that lost its position.  When that occurs then the Z would home wherever the nozzle happened to be in the X Y and the nozzle could hit the bed.

Georgi65, 2023-10-26:
> I think that's it. I'm not sure what conditions would have to happen for the Z to be the only stepper that lost its position. When that occurs then the Z would home wherever the nozzle happened to be in the X Y and the nozzle could hit the bed.

@mdalamond Have you tested this G28 O ;home all untrusted axes problem? 
I have the same problem I was super super super scared when bed slammed into the back of the machine. I hope I didnt destroy something? 

Can I copy G code from creality slicer? do Cura? 

G28 ;Home

G92 E0 ;Reset Extruder
G1 Z2.0 F3000 ;Move Z Axis up
G1 X10.1 Y20 Z0.28 F5000.0 ;Move to start position
G1 X10.1 Y200.0 Z0.28 F1500.0 E15 ;Draw the first line
G1 X10.4 Y200.0 Z0.28 F5000.0 ;Move to side a little
G1 X10.4 Y20 Z0.28 F1500.0 E30 ;Draw the second line
G92 E0 ;Reset Extruder
G1 Z2.0 F3000 ;Move Z Axis up

Or I destroy something?

Georgi65, 2023-10-26:
[image]
[image]

There is 5.5.0 Cura update and differance is that "0" should I try to run it for my ender s1 pro? 
Also there is my Gcode from Creality Slicer compared to version with BUG 5.4.0 
Does the order matter?
What exactly do these commands in the blue box do? and why are they not in Creality cfg?

Georgi65, 2023-10-26:
5.3.0 have the same start g-code as my Creality. 
Can I use this code in 5.5.0? 
[image]
@GregValiant

GregValiant, 2023-10-26:
You can use the StartUp gcode from any version of Cura.  If your machine has ABL make sure you have the ABL commands in the startup.
Gcode is very linear...now "Do this" now "Do this" now "Do this" on and on, line by line.  You should look up the gcode command definitions on the [MarlinFW.org](https://marlinfw.org/meta/gcode/) site.  There are maybe half a dozen that get used all the time.  Your new hobby includes programming a 4 axes robot that happens to squirt molten plastic.  You need to have a working idea of what the different gcodes do.

The StartUp and Ending G-Code from any version of Cura will run.  The question is always "what does the printer understand" so the firmware is involved.  99% of the different startups you will see will work.
That "prep surfaces before Auto-Home for better accuracy" is complete BS.  Auto-Home accuracy has everything to do with the repeatability of the endstop switches and has nothing to do with anything else.  ABL machines are a little different and my guess would be that someone adapted an ABL startup and forgot to take that line out.

Here is what I use on my Ender 3 Pro with NO ABL.
;
;StartUp No ABL
;
G90 ;absolute movement
G21 ;metric units
M82 ;absolute extrusion
M220 S100 ;reset the feedrate to 100%
M221 S100 ;reset the flowrate to 100%
M140 S{material_bed_temperature_layer_0} ;start to heat the bed
M109 S{material_print_temperature_layer_0} ;heat the hot end and wait for it
M190 S{material_bed_temperature_layer_0} ;wait for the bed
G92 E0 ; Reset E
G28 ; Home all axes
G1 F2700 Z10.0 ; Move Z
G0 F5000 X1.0 Y20 Z0.3 ; Move
G1 F1500 X1.0 Y200.0 E15 ; First purge line
G0 F5000 X3.0 Y200.0 ; Move
G1 F1500 X3.0 Y20 E30 ; Second purge line
G92 E0 ; Reset Extruder
G1 F1200 Z12.0 ; Move Z Axis up
G4 S1 ; short pause for string to get hard
G0 F5000 X3.0 Y35 Z.35 ; Move over
G0 F5000 X3 Y50 Z0.35 ;wipe
;
;End of StartUp
;

The snippet below would be acceptable for an ABL printer but the actual ABL commands (M420 and G29) may be slightly different or have a couple of parameters.  That "Prep the surface" comment would mean something here because you want the bed at printing temperature when the ABL routine runs as the build surface can expand and move a little bit.  You don't want the nozzle oozing during the ABL routine but it needs to be warmed up as it may expand and get longer when it's hot.  You can see that the nozzle is heating to 150 before the ABL.

;
; StartUp for ABL
;
G90 ;absolute movement
G21 ;metric units
M82 ;absolute extrusion
M220 S100 ;reset the feedrate to 100%
M221 S100 ;reset the flowrate to 100%
M104 S150 ;warm up the hot end but below the oozing temperature
M190 S{material_bed_temperature_layer_0} ;heat the bed and wait for it
G28 ;AutoHome
G0 Z2 ; move up
M420 ;load mesh
G29 ;ABL
G0 Z5 ; move up
G0 X0 Y0 ; move to corner to ooze
M109 S{material_print_temperature_layer_0} ;heat the nozzle to Initial Layer Print Temperature
G92 E0 ; Reset E
G1 F1200 Z10.0 ; Move Z
G0 F5000 X1.0 Y20 Z0.3 ; Move
G1 F1500 X1.0 Y200.0 E15 ; First purge line
G0 F5000 X3.0 Y200.0 ; Move
G1 F1500 X3.0 Y20 E30 ; Second purge line
G92 E0 ; Reset Extruder
G1 F1200 Z12.0 ; Move Z Axis up
G4 S1 ; short pause for string to get hard
G0 F5000 X3.0 Y35 Z.35 ; Move over
G0 F5000 X3 Y50 Z0.35 ;wipe
;
;End of StartUp
;

Georgi65, 2023-10-27:
my s1 pro have CR-Touch is this ABL? My gcode from creality dont have ABL command.

GregValiant, 2023-10-27:
It would work but as I said, you would need to know the correct M420 and G29 lines.  Both settings accept parameters (that adjust the setting).  The light reading below is one of the reasons I don't have ABL.

This is from the Marlin site regarding G29 and what parameters might be accepted.
G29 [A<bool>] [C<bool>] [D<bool>] [E<bool>] [J<bool>] [O] [Q<bool>] [V<0-4>]

- [A<bool>] Abort leveling procedure in-progress (PROBE_MANUALLY)
- [C<bool>] Create fake points for testing. (DEBUG_LEVELING_FEATURE)
- [D<bool>] Dry-Run mode. Just probe the grid but don’t update the bed leveling data
- [E<bool>] By default G29 will engage the Z probe, test the bed, then disengage.  Include “E” to engage/disengage the Z probe for each sample.  There’s no extra effect if you have a fixed Z probe. (without PROBE_MANUALLY)
- [J<bool>] Jettison the leveling data stored in SRAM and turn off leveling compensation. Data in EEPROM is not affected.
- [O] Optional. If leveling is already enabled then exit without leveling. (1.1.9)
- [Q<bool>] Query the current leveling state (PROBE_MANUALLY, DEBUG_LEVELING_FEATURE)
- [V<0-4>]	Set the verbose level

This is M420:
M420 [C<bool>] [L<int>] [S<bool>] [T<0|1|4>] [V<bool>] [Z<linear>]

- [C<bool>] Center the mesh on the mean of the lowest and highest points
- [L<int>] Load mesh from EEPROM index (Requires AUTO_BED_LEVELING_UBL and EEPROM_SETTINGS)
- [S<bool>] Set enabled or disabled. A valid mesh is required to enable bed leveling. If the mesh is invalid / incomplete leveling will not be enabled.
- [T<0|1|4>] Format to print the mesh data
-- T0: Human readable
-- T1: CSV
-- T4: Compact
- [V<bool>] Verbose: Print the stored mesh / matrix data

- [Z<linear>] Set Z fade height (Requires ENABLE_LEVELING_FADE_HEIGHT)
-- With Fade enabled, bed leveling correction is gradually reduced as the nozzle gets closer to the Fade height. Above the Fade height no bed leveling compensation is applied at all, so movement is machine true.
Set to 0 to disable fade, and leveling compensation will be fully applied to all layers of the print.