Ultimaker/Cura #16011: 5.3.1 profiles not appearing/importing into 5.4.0 stable

EoinLX, 2023-07-05:
### Application Version

5.4.0

### Operating System

Windows 11

### Printer

Elegoo 3 Plus & Creality 5 pro

### Reproduction steps

Upgraded to 5.4.0
Noticed profiles weren't there.
Exported from 5.3.1
Attempted import into 5.4.0 was unsuccessful.

### Actual results

Results are that all my profiles for both printers won't import to 5.4

### Expected results

That the profiles should import or carry over from previous versions.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[5.3.1 cura.log](https://github.com/Ultimaker/Cura/files/11956635/5.3.1.cura.log)
[5.4.0 cura.log](https://github.com/Ultimaker/Cura/files/11956637/5.4.0.cura.log)

Jo-Con-El, 2023-07-30:
I'm having the same problem.  I can upload a zip with my profiles from 5.3 so you can see that the upgrade is not working.

karol-wojcik, 2023-08-04:
Same thing on my end

MariMakes, 2023-08-08:
Hey @EoinLX,

Welcome to the UltiMaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

We've seen more reports of something wrong with the Elegoo Profiles upon installing. 
We are looking into this, but we are having some trouble finding what is happening.

This seems to be a duplicate of https://github.com/Ultimaker/Cura/issues/16007
There are instructions for a workaround to get unstuck for now. 

I'll be closing this issue as a duplicate but you can follow the progress here: https://github.com/Ultimaker/Cura/issues/16007

closed 2023-08-08 by MariMakes