Ultimaker/Cura #16018: hole sliced only in 90 degree. flat on bed not

369Martin369, 2023-07-05:
### Cura Version

5.3.1

### Operating System

Windows 10

### Printer

Anycubic i3 mega

### Describe model location

flat failed, 90 degrees okay

### Describe your model

with 4.13.1 no problem

### Add your .zip here ⬇️

[Panelhalterv9.zip](https://github.com/Ultimaker/Cura/files/11961483/Panelhalterv9.zip)

Asterchades, 2023-07-06:
Your model has some faults but they don't appear to be sufficient to cause the results you're (briefly) describing. However due to the absence of any screen captures accurately showing the issue, or a project file which includes the profile used to experience the issue, further diagnostics is going to be almost impossible.

Best guess at this point is you have the `Remove All Holes` option enabled under Mesh Fixes. When the model is vertical this would do nothing, but laying down these would be filled in by the setting. Under normal circumstances this setting should be disabled.

Should that not help, please upload a project file containing the model in a state where the issue is present.

369Martin369, 2023-07-13:
Remove holes not activated.
Stl is directly from fusion 360 created