Ultimaker/Cura #16019: Fail to load STL

grogoreo, 2023-07-05:
### Cura Version

5.4.0

### Operating System

MacOS Ventura 13.4.1

### Printer

Creality CR6 SE

### Reproduction steps

1. Fresh install of Cura (also happens in previous versions)
2. Cura runs fine
3. Try to open STL

### Actual results

File doesn't load.

File has correct access permissions.

From log:
2023-07-05 20:35:24,060 - INFO - [MainThread] cura.CuraApplication.readLocalFile [1779]: Attempting to read file file:///Users/grogoreo/Downloads/SPT_Smooth.stl

 No further information.

### Expected results

### Workaround

1. Removed configuration files in Library/Application\ Support/cura
2. Complete new configuration and choose ONLY the Ultimaker S5
3. Open the STL and it opens, placing the object on the virtual printer bed
4. Add any additional printer and select
5. As long as the model isn't manipulated or options changed in anyway Cura sometimes slices it
6. If Cura quits with the none Ultimaker S5 printer selected Cura won't load the STL file. As long as Ultimaker S5 was selected and then changed to the alternative printer the model remains but is unworkable as before.

### Add your .zip and screenshots here ⬇️

[UMcura.log](https://github.com/Ultimaker/Cura/files/11961491/UMcura.log)
[CRcura.log](https://github.com/Ultimaker/Cura/files/11961492/CRcura.log)

MariMakes, 2023-08-08:
Hey Hey @grogoreo,

Welcome to the UltiMaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

We've received some reports with the shipped STL reader not working for everyone. 
Could it be that our STL reader plug-in is disabled by accident?
If you go to the marketplace, click on the gear, and scroll down until you see STL Reader, you should be able to Enable it again from there.
[image]

Can you let us know if that is your issue?
If that's not the case, just let us know we'll take another look. 👍
Than it might be a duplicate of https://github.com/Ultimaker/Cura/issues/16249

grogoreo, 2023-08-08:
Thank you for getting back to me.

The STL reader was loaded before because I was able to load an STL if I
chose the Ultimaker S5 but if I made any changes to the position or
rotation it wouldn't slice. I disabled the plugin and re-enabled but no
difference.

The less than ideal solution from @dnbernier to disable USB printing has
worked and am able to do everything without issue (
https://github.com/Ultimaker/Cura/issues/16249#issuecomment-1654809846)

MariMakes, 2023-08-09:
Hey @grogoreo,

Thanks for the update!

I'm glad to hear you are not stuck anymore, but disabling the USB Printing Plug-in, is indeed only a temporary workaround but a super helpful in finding a solution.

I'll bring it up to the team to see what they can do to improve it, we have a number of reports coming in on this subject. 
Fingers crossed 🤞

grogoreo, 2023-08-09:
That sounds great, thanks for looking into it